alibaba/tidevice #4: 对于多台手机启动wda，怎么实现呢，没有指定端口操作吗

cassie9, 2021-01-26:


codeskyblue, 2021-01-27:
举个例子，比如你有两台手机，udid分别是 u1, u2

先启动 `tidevice -u u1 xctest` 然后再启动 `tidevice -u u2 xctest`
8100端口是在手机上监听的，所以不会冲突。

连接的时候需要用iproxy将手机内的8100映射到pc上

`iproxy -u u1 8100 9100`  通过9100端口连接到u1
`iproxy -u u2 8100 9200` 通过9200端口连接到u2

Louis-me, 2021-01-27:
-关注中，谢谢

cassie9, 2021-01-27:
> 举个例子，比如你有两台手机，udid分别是 u1, u2
> 
> 先启动 `tidevice -u u1 xctest` 然后再启动 `tidevice -u u2 xctest`
> 8100端口是在手机上监听的，所以不会冲突。
> 
> 连接的时候需要用iproxy将手机内的8100映射到pc上
> 
> `iproxy -u u1 8100 9100` 通过9100端口连接到u1
> `iproxy -u u2 8100 9200` 通过9200端口连接到u2

get，谢谢

jeniva, 2021-01-29:
请教一下Windows上怎么执行iproxy -u u1 8100 9100呢？需要安装什么？

electricbubble, 2021-01-29:
> https://iphonedevwiki.net/index.php/SSH_Over_USB
## SSH over USB using usbmuxd
You can either download a binary and run that or use a python script. The python script is a lot slower than the binary version. On Linux the python method is mostly deprecated, use the binary version provided by libimobiledevice. There is also a newer solution called gandalf.

### Using binary
On `Windows`, ensure iTunes is installed, then download `itunnel_mux_rev71.zip` from [Google Code](https://code.google.com/archive/p/iphonetunnel-usbmuxconnectbyport/downloads). Unzip to a directory of choice.

On `OS X` and `Linux`, install [usbmuxd](https://github.com/libimobiledevice/usbmuxd) from your package manager.

Then:
- Windows: Run `path/to/itunnel_mux.exe --iport 22 --lport 2222`
- OS X/Linux: `iproxy 2222 22`

jeniva, 2021-01-29:
[image]
我之前也试过这个，就是没成功，用法是这样吗？

electricbubble, 2021-01-29:
确保 `iTunes` 已安装
> On Windows, ensure `iTunes` is installed

jeniva, 2021-01-29:
好了，谢谢！现在有个问题就是我用这个写airtest的自动化，start_app会报错，其他的都是好的，这个知道是什么原因吗？
[image]

electricbubble, 2021-01-29:
根据 `Airtest` 的代码来看, 请求给 `WDA` 的响应都会判断 `status` 是否为 `0`
```python
def httpdo(url, method='GET', data=None):
    """
    Do HTTP Request
    """
    start = time.time()
    if isinstance(data, dict):
        data = json.dumps(data)
    if DEBUG:
        print("Shell: curl -X {method} -d '{data}' '{url}'".format(method=method.upper(), data=data or '', url=url))

    try:
        response = requests.request(method, url, data=data, timeout=HTTP_TIMEOUT)
    except (requests.exceptions.ConnectionError, requests.exceptions.ReadTimeout) as e:
        # retry again
        print('retry to connect, error: {}'.format(e))
        time.sleep(1.0)
        response = requests.request(method, url, data=data, timeout=HTTP_TIMEOUT)

    retjson = response.json()
    if DEBUG:
        ms = (time.time() - start) * 1000
        print('Return ({:.0f}ms): {}'.format(ms, json.dumps(retjson, indent=4)))
    r = convert(retjson)
    if r.status != 0:
        raise WDAError(r.status, r.value)
    return r
```
> https://github.com/AirtestProject/Airtest/blob/6195988071e10257b1aae57cb95782f26c7d617e/airtest/core/ios/wda_client.py#L59-L84

我觉得你应该找 [AirtestProject/Airtest](https://github.com/AirtestProject/Airtest) 反馈下

jeniva, 2021-01-29:
好的，谢谢啦，我也可以用tidevice launch替代使用，这个停止app有方法的吗？

jongyu, 2021-01-31:
> 请教一下Windows上怎么执行iproxy -u u1 8100 9100呢？需要安装什么？
用libimobiledeviceWindows版本的，里面包含iproxy
https://github.com/libimobiledevice-win32/imobiledevice-net/releases

yimelia, 2021-02-01:
> 好了，谢谢！现在有个问题就是我用这个写airtest的自动化，start_app会报错，其他的都是好的，这个知道是什么原因吗？
> [image]

airtest后续会把兼容工作跟上的，可以等待后续版本的更新

jiaxw32, 2021-02-02:
macOS 亲测好用

1.  `iproxy` 转发端口
```shell
# 8101、8102 为本地端口，8100 为设备中 wda 监听端口
iproxy 8101 8100 -u <deviceid_01> &
iproxy 8102 8100 -u <deviceid_02> &
```
2. `tidevice` 启动 wda
```
tidevice -u <deviceid_01> xctest -B com.facebook.WebDriverAgentRunner.xctrunner &
tidevice -u <deviceid_02> xctest -B com.facebook.WebDriverAgentRunner.xctrunner &
```
3. 查看 `wda` 连接状态
```shell
curl -X GET -H 'cache-control: no-cache' -H 'Content-Type: application/json' --url "http://localhost:8101/status";
curl -X GET -H 'cache-control: no-cache' -H 'Content-Type: application/json' --url "http://localhost:8102/status"
```

mp55, 2021-02-07:
> [image]
> 我之前也试过这个，就是没成功，用法是这样吗？

你这个问题怎么解决的呀，我装了itunes还是报这个错

codeskyblue, 2021-02-13:
@mp55 试试新增加的命令 tidevice relay 8100 8100

mp55, 2021-02-18:
新的可以用 谢谢

FuChen1649, 2021-07-09:
你这个岂不是天生就可以多机并发了。。之前搞macaca起多个server 老痛苦了。。

closed 2023-10-19 by codeskyblue