alibaba/tidevice #7: tidevice.exceptions.MuxError: socket unix:/var/run/usbmuxd unable to connect

li8911255, 2021-01-27:
在 Ubuntu 上执行 tidevice list 会报异常

系统：5.4.0-60-generic #67~18.04.1-Ubuntu
具体日志：
List of apple devices attached
Traceback (most recent call last):
  File "/root/.pyenv/versions/3.8.0/bin/tidevice", line 10, in <module>
    sys.exit(main())
  File "/root/.pyenv/versions/3.8.0/lib/python3.8/site-packages/tidevice/__main__.py", line 407, in main
    actions[args.subparser](args)
  File "/root/.pyenv/versions/3.8.0/lib/python3.8/site-packages/tidevice/__main__.py", line 81, in cmd_list
    for dinfo in um.device_list():
  File "/root/.pyenv/versions/3.8.0/lib/python3.8/site-packages/tidevice/_usbmux.py", line 72, in device_list
    data = self.send_recv(payload)
  File "/root/.pyenv/versions/3.8.0/lib/python3.8/site-packages/tidevice/_usbmux.py", line 45, in send_recv
    with self.create_connection() as s:
  File "/root/.pyenv/versions/3.8.0/lib/python3.8/site-packages/tidevice/_usbmux.py", line 42, in create_connection
    return PlistSocket(self.__address, self._next_tag())
  File "/root/.pyenv/versions/3.8.0/lib/python3.8/site-packages/tidevice/_safe_socket.py", line 89, in __init__
    super().__init__(addr)
  File "/root/.pyenv/versions/3.8.0/lib/python3.8/site-packages/tidevice/_safe_socket.py", line 44, in __init__
    raise MuxError("socket unix:{} unable to connect".format(addr))
tidevice.exceptions.MuxError: socket unix:/var/run/usbmuxd unable to connect

electricbubble, 2021-01-27:
根据报错信息 `tidevice.exceptions.MuxError: socket unix:/var/run/usbmuxd unable to connect`

你的 `Ubuntu` 上应该没有安装 `usbmuxd`

 
https://github.com/alibaba/taobao-iphone-device/blob/af697dea3b9f6bbb208e5126163dfc52019dc85a/tidevice/_safe_socket.py#L36-L44

codeskyblue, 2021-01-27:
According to this website: https://askubuntu.com/questions/812006/how-can-i-mount-my-iphone-6s-on-ubuntu-16-04

you need to install usbmuxd with

`sudo apt-get install usbmuxd`

electricbubble, 2021-01-27:
or 👇
https://github.com/libimobiledevice/usbmuxd#debian--ubuntu-linux

closed 2021-01-28 by li8911255

jeniva, 2021-03-31:
我这个usbmuxd应该是装成功了吧，为啥还是提示这个错误呢？
[image]

codeskyblue, 2021-04-01:
你这个属于装上了，但是没启动吧

jeniva, 2021-04-01:
usbmuxd这个还需要怎么启动？

thinkhy, 2021-12-05:
sudo systemctl start usbmuxd

Linlt1, 2022-08-18:
Cenos 如何安装这个玩意呢？ @codeskyblue

chenjigang4167, 2022-11-29:
> sudo systemctl start usbmuxd

我的是centos7系统，通过systemctl restart usbmuxd命令启动usbmuxd没反应，直接执行usbmuxd，提示“usbmuxd: ERROR: Child process exited with error 1, check syslog for messages.”，所以usbmuxd一直都是处于未启动的状态，运行tidevice list就会报错，也就连接不了手机。。。请问一下这种该如何解决呢？