alibaba/tidevice #8: the command: tidevice --socket is not in effect

iTestinglab, 2021-01-27:
When I execute the cmd:  tidevice --socket 127.0.0.1:8100,  it is not in effect
[image]

electricbubble, 2021-01-27:
https://github.com/alibaba/taobao-iphone-device/blob/af697dea3b9f6bbb208e5126163dfc52019dc85a/tidevice/__main__.py#L376
> usbmuxd listen address, host:port or local-path

should use 
- linux or darwin 
  `/var/run/usbmuxd`
- windows
  `127.0.0.1:27015`

iTestinglab, 2021-01-27:
> https://github.com/alibaba/taobao-iphone-device/blob/af697dea3b9f6bbb208e5126163dfc52019dc85a/tidevice/__main__.py#L376
> 
> > usbmuxd listen address, host:port or local-path
> 
> should use
> 
> * linux or darwin
>   `/var/run/usbmuxd`
> * windows
>   `127.0.0.1:27015`

my env is darwin,   I means the command prompt is invalid

iTestinglab, 2021-01-27:
> https://github.com/alibaba/taobao-iphone-device/blob/af697dea3b9f6bbb208e5126163dfc52019dc85a/tidevice/__main__.py#L376
> 
> > usbmuxd listen address, host:port or local-path
> 
> should use
> 
> * linux or darwin
>   `/var/run/usbmuxd`
> * windows
>   `127.0.0.1:27015`

my env is darwin, I means the command  is invalid：`tidevice --socket 127.0.0.1:8100`

electricbubble, 2021-01-27:
1. `8100` is not a listening port for `usbmuxd`
2. `127.0.0.1:27015` only works on `windows`, because there is no `/var/run/usbmuxd` in `windows`

> `tidevice` does not work through port `8100` of `wda`

electricbubble, 2021-01-28:
Maybe you just want to see an error message like this? 👇

`tidevice --socket 127.0.0.1:8100 list`

```bash
(venv) ➜  tidevice --socket 127.0.0.1:8100 list
List of apple devices attached
Traceback (most recent call last):
  File "/Users/_/Documents/Workspace/Python/test/venv/bin/tidevice", line 10, in <module>
    sys.exit(main())
  File "/Users/_/Documents/Workspace/Python/test/venv/lib/python3.6/site-packages/tidevice/__main__.py", line 407, in main
    actions[args.subparser](args)
  File "/Users/_/Documents/Workspace/Python/test/venv/lib/python3.6/site-packages/tidevice/__main__.py", line 81, in cmd_list
    for dinfo in um.device_list():
  File "/Users/_/Documents/Workspace/Python/test/venv/lib/python3.6/site-packages/tidevice/_usbmux.py", line 72, in device_list
    data = self.send_recv(payload)
  File "/Users/_/Documents/Workspace/Python/test/venv/lib/python3.6/site-packages/tidevice/_usbmux.py", line 45, in send_recv
    with self.create_connection() as s:
  File "/Users/_/Documents/Workspace/Python/test/venv/lib/python3.6/site-packages/tidevice/_usbmux.py", line 42, in create_connection
    return PlistSocket(self.__address, self._next_tag())
  File "/Users/_/Documents/Workspace/Python/test/venv/lib/python3.6/site-packages/tidevice/_safe_socket.py", line 89, in __init__
    super().__init__(addr)
  File "/Users/_/Documents/Workspace/Python/test/venv/lib/python3.6/site-packages/tidevice/_safe_socket.py", line 48, in __init__
    self._sock.connect(addr)
ConnectionRefusedError: [Errno 61] Connection refused

```

closed 2021-01-28 by iTestinglab