alibaba/tidevice #10: 期望后面支持打印进程log

canfeng11, 2021-01-28:


codeskyblue, 2021-01-28:
类似Mac的Console 吗？

canfeng11, 2021-01-28:
是的，mac的控制台难用的一言难尽，想找个替代的，貌似都没有，使用libimobiledevice工具里面提供的工具查看log的，貌似也是各种不顺利，期望这个工具能够查看log

codeskyblue, 2021-01-28:
我之前都是一直在用 自带的那个Console.app, 感觉还行。你可以分享下你的一些 idevicesyslog的使用经验吗？ 我看看怎么在 tidevice中改进

canfeng11, 2021-01-29:
全是坑：
1、idevicesyslog 首先安装启动是个问题，当每次苹果系统大版本更新的时候，你需要更新command tools，更新完成之后还要用最新的代码编译构建
2、使用过程中感觉也有问题，用着用着会断开，不知道啥问题

electricbubble, 2021-02-04:
```python
from tidevice import _usbmux
from tidevice import Device

if __name__ == '__main__':
    um = _usbmux.Usbmux()
    devList = um.device_list()
    for dev in devList:
        print(dev)

    dev = Device(devList[0]["SerialNumber"], um)
    logServiceApi = dev._unsafe_start_service("com.apple.syslog_relay")
    while True:
        buf = logServiceApi.recvall(512)
        print(buf.decode(), end="")

```

codeskyblue, 2021-02-04:
> ```python
> from tidevice import _usbmux
> from tidevice import Device
> 
> if __name__ == '__main__':
>     um = _usbmux.Usbmux()
>     devList = um.device_list()
>     for dev in devList:
>         print(dev)
> 
>     dev = Device(devList[0]["SerialNumber"], um)
>     logService = dev._unsafe_start_service("com.apple.syslog_relay")
>     while True:
>         buf = logService.recvall(512)
>         print(buf.decode(), end="")
> ```

你怎么这么优秀

electricbubble, 2021-02-04:
😂看了好几天源码了。。

codeskyblue, 2021-02-04:
@electricbubble 看来在新公司混的挺开的嘛

electricbubble, 2021-02-04:
> @electricbubble 看来在新公司混的挺开的嘛

😂正好这几天偷摸着看的。。

codeskyblue, 2021-02-04:
https://github.com/alibaba/taobao-iphone-device/commit/2b38988cd0bdef3c0579c1a4b930de8b45c67727  随便写写

codeskyblue, 2021-02-04:
@electricbubble 上面那个 crashlog你要不要也搞搞

electricbubble, 2021-02-04:
> @electricbubble 上面那个 crashlog你要不要也搞搞

😥我还不晓得哪些情况会产生 crash log。。。
需要先认识认识这日志长啥样子😂

closed 2021-03-02 by codeskyblue