alibaba/tidevice #9: tidevice.exceptions.MuxReplyError: UsbmuxReplyCode.BadDevice

li8911255, 2021-01-28:
在一台 iPhone 7P 设备上运行 tidevice applist 报错，其他设备正常

具体错误日志

tidevice applist
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/tidevice/_device.py", line 192, in handshake
    self._pair_record = self._read_pair_record()
  File "/usr/local/lib/python3.8/site-packages/tidevice/_device.py", line 163, in _read_pair_record
    data = self._usbmux.send_recv(payload)
  File "/usr/local/lib/python3.8/site-packages/tidevice/_usbmux.py", line 48, in send_recv
    self._check(recv_data)
  File "/usr/local/lib/python3.8/site-packages/tidevice/_usbmux.py", line 81, in _check
    raise MuxReplyError(data['Number'])
tidevice.exceptions.MuxReplyError: UsbmuxReplyCode.BadDevice

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/tidevice", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/site-packages/tidevice/__main__.py", line 407, in main
    actions[args.subparser](args)
  File "/usr/local/lib/python3.8/site-packages/tidevice/__main__.py", line 226, in cmd_applist
    for info in d.installation.iter_installed():
  File "/usr/local/lib/python3.8/site-packages/tidevice/_device.py", line 483, in installation
    conn = self.start_service(Installation.SERVICE_NAME)
  File "/usr/local/lib/python3.8/site-packages/tidevice/_device.py", line 400, in start_service
    return self._unsafe_start_service(name)
  File "/usr/local/lib/python3.8/site-packages/tidevice/_device.py", line 408, in _unsafe_start_service
    with self.create_session() as s:
  File "/usr/local/Cellar/python@3.8/3.8.5/Frameworks/Python.framework/Versions/3.8/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/usr/local/lib/python3.8/site-packages/tidevice/_device.py", line 272, in create_session
    "HostID": self.pair_record['HostID'],
  File "/usr/local/lib/python3.8/site-packages/tidevice/_device.py", line 127, in pair_record
    self.handshake()
  File "/usr/local/lib/python3.8/site-packages/tidevice/_device.py", line 195, in handshake
    self.pair()
  File "/usr/local/lib/python3.8/site-packages/tidevice/_device.py", line 180, in pair
    iconsole_path = get_binary_by_name("iconsole")
  File "/usr/local/lib/python3.8/site-packages/tidevice/_utils.py", line 47, in get_binary_by_name
    raise RuntimeError("Binary file {} not exist".format(name))
RuntimeError: Binary file iconsole not exist

codeskyblue, 2021-01-28:
设备没配对，配对一下就好了

codeskyblue, 2021-01-28:
本来是可以实现pair这个功能的，但是pair依赖 pyopenssl这个库，而这个库又不是太好安装。

li8911255, 2021-01-28:
重新信任一下就好了

closed 2021-01-28 by li8911255