alibaba/tidevice #12: 找不到iOS14.4.zip文件

jongyu, 2021-01-31:
requests.exceptions.HTTPError: 404 Client Error: Not Found for url: https://tool.appetizer.io/iGhibli/iOS-DeviceSupport/raw/master/DeviceSupport/14.4.zip
requests.exceptions.HTTPError: 404 Client Error: Not Found for url: https://github.com/iGhibli/iOS-DeviceSupport/raw/master/DeviceSupport/14.4.zip
自动webdriverrunner时报找不到文件，下载地址里面没有对应文件，但是里面的iOS14文件名包含其他的内容，我感觉你们时没有测iOS14的设备？
https://tool.appetizer.io/iGhibli/iOS-DeviceSupport/tree/master/DeviceSupport

jongyu, 2021-01-31:
[image]

jongyu, 2021-01-31:
[image]

codeskyblue, 2021-01-31:
fixed in tidevice 0.1.4

closed 2021-01-31 by codeskyblue

electricbubble, 2021-01-31:
😭 刚 `fork` 想提交蹭一个贡献者, 就被修复了

codeskyblue, 2021-01-31:
> 😭 刚 `fork` 想提交蹭一个贡献者, 就被修复了

There are still many chances

jongyu, 2021-01-31:
大佬速度快得很嘛，我看你源码后，我自己通过Mac上的镜像拷贝过去了。项目很不错，如果可以把iproxy工具加进入进去就好了，弄了一天发现，Appium跑不起来，应该是appium有做环境判断，提示`An unknown server-side error occurred while processing the command. Original error: Could not determine Xcode version: Could not find path to Xcode by symlinks located in /var/db/xcode_select_link, or /usr/share/xcode-select/xcode_dir_link` ， 希望大佬给appium那边提个需求，那样就无敌了，我也看了facebook-wda的源码，它是自己调接口启动的xctest，没有像appium做了系统判断

jongyu, 2021-01-31:
对了，windows下我是用`libimobiledevice` Windows版本的 里面包含了`iproxy.exe`工具,可以通过它转发端口
https://github.com/libimobiledevice-win32/imobiledevice-net/releases/

commit afe14aac (2021-01-31): fix #12
commit 138d523f (2021-02-01): fix #12

Former-commit-id: afe14aacf9dcf523282640a9ce863e1aded89231