alibaba/tidevice #14: wda重签名后运行tidevice xctest 报错 socket connection broken

carrypink, 2021-02-02:
python3.6&3.8
mac 10.15.5
log:
```
[I 210201 21:10:10 _device:912] BundleID: com.xxxxxxxxxx.WebDriverAgentRunner-Runner
[I 210201 21:10:10 _device:914] DeviceIdentifier: 00008030-xxxxxxxxxxxxE
[I 210201 21:10:10 _device:775] SignIdentity: 'iPhone Distribution: xxxxxxxxx Co., Ltd.'
Traceback (most recent call last):
  File "/Users/xx/.pyenv/versions/uiauto8/bin/tidevice", line 8, in <module>
    sys.exit(main())
  File "/Users/xx/.pyenv/versions/3.8.6/envs/uiauto8/lib/python3.8/site-packages/tidevice/__main__.py", line 429, in main
    actions[args.subparser](args)
  File "/Users/xx/.pyenv/versions/3.8.6/envs/uiauto8/lib/python3.8/site-packages/tidevice/__main__.py", line 209, in cmd_xctest
    d.xctest(args.bundle_id, logger=setup_logger(level=logging.INFO), env=env)
  File "/Users/xx/.pyenv/versions/3.8.6/envs/uiauto8/lib/python3.8/site-packages/tidevice/_device.py", line 985, in xctest
    pid = self._launch_wda(bundle_id, session_identifier, env=env, logger=logger)
  File "/Users/xx/.pyenv/versions/3.8.6/envs/uiauto8/lib/python3.8/site-packages/tidevice/_device.py", line 786, in _launch_wda
    for fname in fsync.listdir("/tmp"):
  File "/Users/xx/.pyenv/versions/3.8.6/envs/uiauto8/lib/python3.8/site-packages/tidevice/_sync.py", line 106, in listdir
    pkg = self._recv()
  File "/Users/xx/.pyenv/versions/3.8.6/envs/uiauto8/lib/python3.8/site-packages/tidevice/_sync.py", line 77, in _recv
    buf = buf + self.recvall(FHeader.size - len(buf))
  File "/Users/xx/.pyenv/versions/3.8.6/envs/uiauto8/lib/python3.8/site-packages/tidevice/_safe_socket.py", line 55, in recvall
    raise MuxError("socket connection broken")
tidevice.exceptions.MuxError: socket connection broken
```

electricbubble, 2021-02-02:
试一下 #13 之后的版本?
> 个人不确定的猜测

codeskyblue, 2021-02-02:
感觉有点数据线不稳的感觉

carrypink, 2021-02-02:
> 试一下 #13 之后的版本?
> 
> > 未经验证的猜测

刚试了最新版，无论新旧版本，同一台手机，用Xcode Test 自动安装上的wda都可以正常运行 `tidevice xctest`，而用重签名后的wda都报这个错

codeskyblue, 2021-02-02:
> > 试一下 #13 之后的版本?
> > > 未经验证的猜测
> 
> 刚试了最新版，无论新旧版本，同一台手机，用Xcode Test 自动安装上的wda都可以正常运行 `tidevice xctest`，而用重签名后的wda都报这个错

你不会是用企业证书签名的吧

carrypink, 2021-02-03:
是企业证书

codeskyblue, 2021-02-03:
嗯，看样子，企业证书不支持，用开发者证书是可以的

mimers, 2022-03-24:
> 嗯，看样子，企业证书不支持，用开发者证书是可以的

为什么不能支持企业证书呢？具体是什么区别？

codeskyblue, 2022-05-09:
企业证书里面有个 task flow 默认是关闭掉的。

closed 2022-05-09 by codeskyblue

Zhangyu754178, 2023-01-03:
我用的个人证书也会出现这个问题，两个不同的证书一个重签后可以正常跑，一个重签名后不能正常跑，出现相同的问题，是要设置什么吗

myungkyojung, 2023-01-17:
> 我用的个人证书也会出现这个问题，两个不同的证书一个重签后可以正常跑，一个重签名后不能正常跑，出现相同的问题，是要设置什么吗

what kind of certificate you have ?
enterprice certificate for distribution?
you got any solutions?

myungkyojung, 2023-01-17:
> 
do you have any solutions??