alibaba/tidevice #16: 多设备使用relay代理给Appium跑自动化会出现xctrunner quited

jongyu, 2021-02-03:
报错信息
```
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/threading.py", line 932, in _bootstrap_inner
[W 210203 13:54:59 _device:1042] xctrunner quited
```
启动代理的代码`tidevice -u ${udid} relay ${extendPort} 8100`
有在Appium中有设置`webDriverAgentUrl`

jongyu, 2021-02-03:
``` 
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/threading.py", line 932, in _bootstrap_inner
[W 210203 14:25:42 _device:1042] xctrunner quited
Error connecting to device: No such file or directory
```

codeskyblue, 2021-02-03:
太抽象了，看不出来原因

electricbubble, 2021-02-03:
确定不是误关闭了 `tidevice relay` ?

https://github.com/alibaba/taobao-iphone-device/blob/efa085f3b1111335ef5508d82a302e168fb698fc/tidevice/_device.py#L1038-L1042

jongyu, 2021-02-03:
> 确定不是误关闭了`tidevice relay`吗？
> 
> https://github.com/alibaba/taobao-iphone-device/blob/efa085f3b1111335ef5508d82a302e168fb698fc/tidevice/_device.py#L1038-L1042

这个意思是tidevice xctest的时候关闭了吗

electricbubble, 2021-02-03:
对, 我贴错了, 我猜的是 `tidevice xctest` 中途被关闭了

jongyu, 2021-02-03:
> 对, 我贴错了, 我猜的是 `tidevice xctest` 中途被关闭了

那可能是运行`tidevice xctest`的时候不稳定造成的

jongyu, 2021-02-03:
用了最新的版本，不会出现代理没了的问题，但还是有遇到xctest断了，tidevice自动重启后，Appium连接之前对的会提示session不存在了
```
[E 210203 19:46:06 _relay:46] connect to device error: UsbmuxReplyCode.ConnectionRefused
```
- 运行xctest的命令
```
tidevice -u ${device.udid} wdaproxy --port ${wda.port}
```

closed 2021-03-18 by jongyu