alibaba/tidevice #19: 求解connect to device error: UsbmuxReplyCode.ConnectionRefused

mp55, 2021-02-07:
wda装好了，连mac 执行iproxy 8100 8100 访问http://localhost:8100/status可以看到手机信息

但是连windosw，执行tidevice xctest -B com.facebook.wda.WebDriverAgent.Runner -e USB_PORT:8200 提示WebDriverAgent start successfully，然后执行tidevice relay 8100 8100，访问http://localhost:8100/status 报错
handle stream from: ('::1', 59494, 0, 0)
connect to device error: UsbmuxReplyCode.ConnectionRefused
请问怎么解呀

metoto, 2021-02-07:
和你一样，看看有没有人知道

electricbubble, 2021-02-07:
> `Windows`
> 确保 `iTunes` 已安装
> 确保 `iOS设备` 已安装 `WDA`

 1. 运行 `WebDriverAgent`
    ```bash
    tidevice xctest -B com.facebook.wda.WebDriverAgent.Runner
    ```
 2. 转发请求到手机，类似于 `iProxy`
    ```bash
    tidevice relay 8100 8100
    ```

---

或者使用新版本的新命令, 效果等同上面
```bash
tidevice wdaproxy -B com.facebook.wda.WebDriverAgent.Runner --port 8100
```
> https://github.com/alibaba/taobao-iphone-device#%E8%BF%90%E8%A1%8Cwebdriveragent

mp55, 2021-02-07:
iTunes版本 12.11.0.26
ios WDA mac上可以用
windows10

操作步骤：
1.执行tidevice xctest -B com.facebook.WebDriverAdsgentRner.xctrunner     包名我改过
2.tidevice relay 8100 8100
3.浏览器访问：http://localhost:8100/status
4.报错

新命令也一样的错，重连的时候还会报wda started failed

mp55, 2021-02-07:
[](https://github.com/mp55/test/blob/main/1.png)

mp55, 2021-02-07:
https://github.com/mp55/test/blob/main/1.png

electricbubble, 2021-02-07:
> https://github.com/mp55/test/blob/main/1.png

连 `macOS` 和 `Windows` 是同一台 `iOS设备` 吗?

看这个报错信息, `WDA` 应该是启动成功了, 
转发服务也工作了, 
但是 `WDA` 默认的 `8100` 端口连不上

---

还是说, 你 `Windows` 有同时连接多个 `iOS设备` ?

mp55, 2021-02-07:
是同一台， 就一个IOS

electricbubble, 2021-02-07:
我暂时连不上 `Windows` 环境...

你可以打开转发服务的 `DEBUG`, 贴下执行日志
```bash
tidevice relay -x 8100 8100
```

mp55, 2021-02-07:
是这样吗，好像没有别的日志
[image]

还有我在这里看到https://github.com/alibaba/taobao-iphone-device/issues/4
[image]
这个错我也报也，看着是说itunes装了就会好的嘛？ 但是我装了也，就是那个同步手机数据的软件吧，itunes需要什么特别的配置么难道

electricbubble, 2021-02-07:
🌚那试试 重装大法
Windows 再安装一次 iTunes

jongyu, 2021-02-07:
> 是这样吗，好像没有别的日志
> [image](https://github.com/mp55/test/blob/main/2.png)
> 
> 还有我在这里看到 #4
> 
> [image]
> 这个错我也报也，看着是说itunes装了就会好的嘛？ 但是我装了也，就是那个同步手机数据的软件吧，itunes需要什么特别的配置么难道

你咋还用这个啊，大佬都直接提供了`tidevice wdaproxy --port 8100`，直接代理出去，你们用都itunnel本来就有问题的吧，都不知道多少年没更新过了

mp55, 2021-02-07:
我就是这个新的成功不了。。就去翻旧的了- -， 那个我重装了也不行。。

electricbubble, 2021-02-07:
> 我就是这个新的成功不了。。就去翻旧的了- -， 那个我重装了也不行。。
> https://github.com/mp55/test/blob/main/3.png?raw=true

我看你的执行路径是在 `C:` 路径下
尝试移到 `非系统盘` 路径下执行看看?
或者 `管理员权限` 执行 `tidevice` 试试?

metoto, 2021-02-08:
使用wdaproxy命令是好的，可以打开status.  但是很奇怪sessionId是null，然后用airtest连接不上，真的很奇怪

{
  "value" : {
    "message" : "WebDriverAgent is ready to accept commands",
    "state" : "success",
    "os" : {
      "testmanagerdVersion" : 28,
      "name" : "iOS",
      "sdkVersion" : "14.2",
      "version" : "13.6.1"
    },
    "ios" : {
      "ip" : "192.168.20.136"
    },
    "ready" : true,
    "build" : {
      "time" : "Feb  6 2021 17:03:42",
      "productBundleIdentifier" : "com.facebook.WebDriverAgentRunner"
    }
  },
  "sessionId" : null
}

electricbubble, 2021-02-08:
1. https://github.com/appium/WebDriverAgent/issues/353

2. Airtest 连不上的信息描述不足

mp55, 2021-02-08:
[image](https://github.com/mp55/test/blob/main/4.png)
D盘也不行。。话说不需要IOS越狱的吧 ？

electricbubble, 2021-02-08:
> [image](https://github.com/mp55/test/blob/main/4.png)
> D盘也不行。。话说不需要IOS越狱的吧 ？

不需要 `越狱`
`管理员权限` 试过了吗?

metoto, 2021-02-08:
> 1. [appium/WebDriverAgent#353](https://github.com/appium/WebDriverAgent/issues/353)
> 2. Airtest 连不上的信息描述不足

airtest的提示如下：
retry to connect, error: HTTPConnectionPool(host='192.168.20.136', port=8100): Max retries exceeded with url: /status (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x0000028FCD737CC0>: Failed to establish a new connection: [WinError 10060] 由于连接方在一段时间后没有正确答复或连接的主机没有反应，连接尝试失败。',))

electricbubble, 2021-02-08:
> [image](https://github.com/mp55/test/blob/main/4.png)
> D盘也不行。。话说不需要IOS越狱的吧 ？

---

我在 `Windows` 测试了下, 无法复现

---

```
connect to device error: UsbmuxReplyCode.ConnectionRefused
```
这个报错只可能是 `WDA` 默认端口 `8100` 未开启的情况

建议你再确认下 `WDA` 的端口有没有被改过

electricbubble, 2021-02-08:
> airtest的提示如下：
> retry to connect, error: HTTPConnectionPool(host='192.168.20.136', port=8100): Max retries exceeded with url: /status (Caused by NewConnectionError('<urllib3.connection.HTTPConnection [...]

---

`192.168.20.136` 这个 `IP` 是 `iOS设备`的?

如果使用了 `tidevice` 的转发
这里的 `IP` 应该是 `localhost` 或 `Windows`/`macOS` 的 `IP`

mp55, 2021-02-08:
我想到一个问题，我mac上每次重新连线 都是重新Build过才能连上的，不重新build就会报[image](https://github.com/mp55/test/blob/main/5.png)
我用过的命令是  xcodebuild -project WebDriverAgent.xcodeproj -scheme WebDriverAgentRunner -destination 'id=00008020-00026D322E9A002E' GCC_TREAT_WARNINGS_AS_ERRORS=0 test
或者   xcodebuild build-for-testing test-without-building -project WebDriverAgent.xcodeproj -scheme WebDriverAgentRunner -destination id=00008020-00026D322E9A002E USE_PORT=8100 test

你们的WAG装完，插拔线是直接
iproxy 8100 8100
http://localhost:8100/status 
就能连上的吗 ？

mp55, 2021-02-08:
> airtest的提示如下：
> retry to connect, error: HTTPConnectionPool(host='192.168.20.136', port=8100): Max retries exceeded with url: /status (Caused by NewConnectionError('<urllib3.connection.HTTPConnection [...]

airtest我是安装官方给的https://github.com/AirtestProject/IOS-Tagent 这个能连，但是也是需要插拔线以后重装才能连上。 他可能需要定制的才行吧 ？

metoto, 2021-02-08:
我这边debug了一下，发现还是session为None导致的airtest的connect_device("ios:///localhost:8100") 报错.
但是 @electricbubble  ，#353 我看已经关闭了，他的master分支上应该没有这个问题才对吧? 还是说这个问题依然存在啊?有没有什么好办法解决？

electricbubble, 2021-02-08:
> 我这边debug了一下，发现还是session为None导致的airtest的connect_device("ios:///localhost:8100") 报错.
> 但是 @electricbubble ，#353 我看已经关闭了，他的master分支上应该没有这个问题才对吧? 还是说这个问题依然存在啊?有没有什么好办法解决？

---

1. `Airtest` 与 [appium/WebDriverAgent](https://github.com/appium/WebDriverAgent) 的兼容性问题, 建议反馈给 [Airtest](https://github.com/AirtestProject/Airtest/issues/new?assignees=&labels=&template=bug_report.md)
2. 自 https://github.com/appium/WebDriverAgent/pull/324 之后, `/status` 都将默认为 `null`
    > `/status` 仅在主动创建 `session` 后, 才会输出 `sessionId`

electricbubble, 2021-02-08:
@metoto 

`body`
```
--> POST http://localhost:8100/session
{"capabilities":{}}
```

mp55, 2021-02-08:
> > [image](https://github.com/mp55/test/blob/main/4.png)
> > D盘也不行。。话说不需要IOS越狱的吧 ？
> 
> 不需要 `越狱`
> `管理员权限` 试过了吗?

管理员试了 不行，换了个电脑也不行，换个手机也不行。你们WDA有不越狱能装的工程或者包不

electricbubble, 2021-02-08:
@mp55 

那基本上只可能是你这台设备的 `WDA` 问题了吧...
我的是直接用 https://github.com/appium/WebDriverAgent

mp55, 2021-02-08:
@electricbubble 
装了你这个就不报错了！  谢谢大佬

closed 2021-02-08 by mp55