alibaba/tidevice #21: 可以直接驱动原生的XCUITest么

LeepengX, 2021-02-07:
不喜欢WDA这种东西，原生的速度和稳定性我感觉也很棒，就是apple推出的基于swift和OC的这些ui自动化
不晓得tidevice能否驱动呢？
感谢
[image]

jongyu, 2021-02-07:
那你干嘛不直接写xctest的测试类，然后在用idb执行

codeskyblue, 2021-02-13:
理论上是可以的，直接也有人想用这个跑IOSMonkey，目前已经完成了改造。你可以试下，然后反馈一下结果

codeskyblue, 2021-03-02:
支持了。我用这个项目 https://github.com/FeiHuang93/XCTest-Demo 测试了一下是可以的。前提是需要手机上已经装上的应用和被测应用

codeskyblue, 2021-03-02:
@LeepengX 你也可以提供个样子工程，我编译后跑跑试试

zhangpeibisha, 2022-03-23:
请问可以支持指定 -only-testing 这样的参数吗～

jongyu, 2022-03-23:
> 可以支持指定-only-testing这样的参数吗～

可以试一下`-e `参数传递， 参考我用[Fastbot_iOS](https://github.com/bytedance/Fastbot_iOS/issues/1)执行的例子

zhangpeibisha, 2022-03-23:
我看了下你的执行命令，tidevice xctest --bundle-id com.xxx.MonkeyQAUITests.xctrunner --target-bundle-id com.xxxx.MonkeyQAUITests -e only-testing:xxxxx/testSetting 我这样去配置后还是不行，这个-e的参数在哪里可以看到说明呀，其实我不知道这个-e可以填写什么参数，还是这个-e其实是单元测试接收的参数值，不是指定运行哪个case

nanoscopic, 2022-04-01:
The excessive complexity and other bugs within WDA are why I wrote a replacement for it, CFA. It is currently closed source though. I will likely be releasing it in the near future, but as a binary to be re-signed that can be used freely.

What is needed is a minimal wrapper around xctest API, and that is what CFA is.

codeskyblue, 2022-04-07:
@nanoscopic  You may ignore CFA. Very look forward for your new WDA alternative released.

The question: "What is needed is a minimal wrapper around xctest API?"  em... I donot known.