alibaba/tidevice #22: [tidevice install error ]io.unsupportedoperation: seek

dhc-zhangshiqiang, 2021-02-08:
commond:
tidevice --udid 648a03dfcbbfda416b28114c04109e07e99xxxxxxx install http://xxxxxxxxx/aaa.ipa

error:
[7.1 MB/s 2s] Traceback (most recent call last):
  File "/home/zsq/.local/bin/tidevice", line 11, in <module>
    sys.exit(main())
  File "/home/zsq/.local/lib/python3.6/site-packages/tidevice/__main__.py", line 475, in main
    actions[args.subparser](args)
  File "/home/zsq/.local/lib/python3.6/site-packages/tidevice/__main__.py", line 121, in cmd_install
    bundle_id = d.app_install(args.filepath_or_url)
  File "/home/zsq/.local/lib/python3.6/site-packages/tidevice/_device.py", line 684, in app_install
    bundle_id = ir.get_bundle_id()
  File "/home/zsq/.local/lib/python3.6/site-packages/tidevice/_ipautil.py", line 57, in get_bundle_id
    return self.get_infoplist()['CFBundleIdentifier']
  File "/home/zsq/.local/lib/python3.6/site-packages/tidevice/_ipautil.py", line 53, in get_infoplist
    return bplist.load(fp)
  File "/home/zsq/.local/lib/python3.6/site-packages/tidevice/plistlib2.py", line 966, in load
    fp.seek(0)
io.UnsupportedOperation: seek

electricbubble, 2021-02-08:
@dhc-zhangshiqiang 

我尝试提交了一个 [PR #23](https://github.com/alibaba/taobao-iphone-device/pull/23)
但是我没有可以用的 `.ipa` 文件
所以, 你可以帮我试一下这段代码, 看下是否可以修复这个问题?

```python
from tidevice import _usbmux
from tidevice import __main__

if __name__ == '__main__':
    um = _usbmux.Usbmux()
    udid = "648a03dfcbbfda416b28114c04109e07e99xxxxxxx"
    __main__.um = um
    d = __main__._udid2device(udid)
    bundle_id = d.app_install("http://xxxxxxxxx/aaa.ipa")
    print(bundle_id)

```

electricbubble, 2021-02-08:
> force tidevice to install py 3.7+
> https://github.com/alibaba/taobao-iphone-device/commit/2662a630e56bc151c076e8ff7f6be9d993e5581d

我觉得你需要通过升级 `Python` 来解决这个问题

codeskyblue, 2021-03-08:
This is because zipfile is seekable after python 3.7
so python 3.6 is not supported

See: https://stackoverflow.com/questions/12821961/seek-a-file-within-a-zip-file-in-python-without-passing-it-to-memory

closed 2021-03-08 by codeskyblue

codeskyblue, 2021-03-08:
python 3.6 我又重新支持了。^_^

commit c500ca26 (2021-03-08): fix python 3.6 support. fix #22