alibaba/tidevice #27: 你好，请问wda的ipa包在哪里下载呢

nuomituan47, 2021-02-18:


codeskyblue, 2021-02-18:
被我删掉了，你可以看这个文档自己打包一下 https://github.com/alibaba/taobao-iphone-device/blob/main/DEVELOP.md

nuomituan47, 2021-02-28:
在window10用tidevice 启动wda报这个错，感觉就差一点什么了，就是搞不定，这是什么情况呢

PS C:\windows\system32> tidevice -u dfdd6fcfd7f5bd4cfc4f6f4953b61eabd6187bfb xctest -B com.apple.test.WebDriverAgentRunner-Runner
[I 210228 18:25:30 _device:919] BundleID: com.apple.test.WebDriverAgentRunner-Runner
[I 210228 18:25:30 _device:921] DeviceIdentifier: dfdd6fcfd7f5bd4cfc4f6f4953b61eabd6187bfb
[I 210228 18:25:31 _device:776] SignIdentity: 'iPhone Developer: nuomituan47@icloud.com (8B4U533N4H)'
[I 210228 18:25:31 _device:782] CFBundleExecutable: WebDriverAgentRunner-Runner
[E 210228 18:25:31 _device:847] Launch failed: NSError(CODE:1 DOMAIN:DTXMessage INFO:{'NSLocalizedDescription': 'The operation couldn’t be completed. Unable to '
                               'launch com.apple.test.WebDriverAgentRunner-Runner '
                               'because it has an invalid code signature, '
                               'inadequate entitlements or its profile has not '
                               'been explicitly trusted by the user. : Failed to '
                               'launch process with bundle identifier '
                               "'com.apple.test.WebDriverAgentRunner-Runner'"})
Traceback (most recent call last):
  File "c:\users\pirate\appdata\local\programs\python\python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\pirate\appdata\local\programs\python\python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\pirate\AppData\Local\Programs\Python\Python39\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "c:\users\pirate\appdata\local\programs\python\python39\lib\site-packages\tidevice\__main__.py", line 483, in main
    actions[args.subparser](args)
  File "c:\users\pirate\appdata\local\programs\python\python39\lib\site-packages\tidevice\__main__.py", line 210, in cmd_xctest
    d.xctest(args.bundle_id, logger=setup_logger(level=logging.INFO), env=env)
  File "c:\users\pirate\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 992, in xcte

nuomituan47, 2021-02-28:
无缘无故解决了，不过wda必须要到xcode里去安装，ipa重签名的没用

codeskyblue, 2021-03-02:
看起来应该是重签名失败

nuomituan47, 2021-03-07:
请问我该如何在puthon里调用tidevice呢
比如：
form tidevice._usbmux import Usbmux
u = Usbmux()
devices = u.device_list()
print(devices)

如果我想在python实现 tidevice -u uuid wdaproxy -B butID --port 8100
这个功能该怎么写呢
我的本意是想打包起来发给没有安装python和tidevice的机器使用

codeskyblue, 2021-03-08:
你看看 __main__.py 中的代码，参考着写一下就好了

weidingxing, 2021-04-12:
加群720547055下载wda.ipa

wjwABCDEFG, 2023-06-25:
> 加群720547055下载wda.ipa

还有吗？求

wjwABCDEFG, 2023-06-25:
> 加群720547055下载wda.ipa

群不在了，还有吗？求