alibaba/tidevice #28: 求解：我启动完wda后，怎么运行appium

pencil1, 2021-02-19:
[image]
[image]

appium的端口要怎么配置，8100肯定不行了，那换成其他的之后，代码里变成8101，又启动不了，有点懵逼

electricbubble, 2021-02-19:
1. `图1`
    使用 `wdaproxy` 命令, 并使用 `--port 8100` 意味着, 你将通过 `PC` 的 `8100` 端口转发到 `WDA`

    所以, 你的 `图2`
      - 主机：`127.0.0.1` 或 `localhost`
      - 端口：`8100`
      
2. `图2`
    报错信息是 [appium-ios-device](https://github.com/appium/appium-ios-device) 的判断逻辑

```js
async function getDefaultSocket (opts = {}) {
  const {
    socketPath = DEFAULT_USBMUXD_SOCKET,
    socketPort = DEFAULT_USBMUXD_PORT,
    socketHost = DEFAULT_USBMUXD_HOST,
    timeout = 5000,
  } = opts;

  let socket;
  if (await fs.exists(socketPath)) {
    socket = net.createConnection(socketPath);
  } else if (process.platform === 'win32'
      || (process.platform === 'linux' && /microsoft/i.test(os.release()))) {
    // Connect to usbmuxd when running on WSL1
    socket = net.createConnection(socketPort, socketHost);
  } else {
    throw new Error(`The usbmuxd socket at '${socketPath}' does not exist or is not accessible`);
  }

  return await new B((resolve, reject) => {
    socket.once('error', reject);
    socket.once('connect', () => resolve(socket));
  }).timeout(timeout);
}
```
> https://github.com/appium/appium-ios-device/blob/b47d5b394f884b597281ebabd48144e6fc17d23f/lib/usbmux/index.js#L64-L87

pencil1, 2021-02-19:
[image]

不是很懂，如果我图2也配8100，appium就直接提示8100占用了~~~所以我用8101是没有错？是 appium-ios-device那个判断问题？

pencil1, 2021-02-19:
[image]
appium这样子配是可以启动的，但是还是报上面的错~~这是需要appium-ios-device这边改才行对吗？

electricbubble, 2021-02-19:
这个报错就很怪了, 你的 `Windows` 是如何跳过这个判断的...
> 我不太熟悉 `Node.js`
```
else if (process.platform === 'win32'
```

pencil1, 2021-02-19:
[image]
我知道了，我的是appium是1.17版本，所以没判断，换成最新的1.20就好了，但是又出现另外一个报错，xcode-select，这命令怎么解？

electricbubble, 2021-02-19:
那就是 [appium-xcode](https://github.com/appium/appium-xcode) 做的事情了 (这模块应该只考虑了 `macOS` 的情况)
> https://github.com/appium/appium-xcode/blob/df580a74a0452444eb68b0c1e19d4d20678dc5d6/lib/xcode.js#L40-L85

electricbubble, 2021-02-19:
如果 `appium` 的客户端可以选择跳过 `appium-xcode` 模块的相关检查,
应该就可以直接使用了
> 未经验证的猜测

pencil1, 2021-02-19:
好的，非常感谢，我有空试试

electricbubble, 2021-02-20:
@pencil1 

你可以试试 `appium` 的这几个 [参数](https://github.com/appium/appium-xcuitest-driver#webdriveragent) 

- `appium:webDriverAgentUrl`: `http://localhost:8100`
- `appium:usePrebuiltWDA`: `true`
- `appium:useXctestrunFile`: `false`

> 同时设置

pencil1, 2021-02-22:
@electricbubble 
[image]
还是有其他报错~

electricbubble, 2021-02-22:
@pencil1 
再加一个参数

> - `appium:webDriverAgentUrl`: `http://localhost:8100`
> - `appium:usePrebuiltWDA`: `true`
> - `appium:useXctestrunFile`: `false`
- `appium:skipLogCapture`: `true`

pencil1, 2021-02-22:
@electricbubble 
[image]
还是这个报错

electricbubble, 2021-02-22:
但是没有异常抛出了,
并且, 底下的日志似乎是成功请求了,
试试 `appium` 的相关操作是否可行?

pencil1, 2021-02-22:
[image]
好奇怪的问题，提示no body，但是用网页打开，是有信息的

electricbubble, 2021-02-22:
1. `with no body` 指的是本次请求不包含 `body`, 不是指 `响应`
2. 再追加一个参数
    > - `appium:webDriverAgentUrl`: `http://localhost:8100`
    > - `appium:usePrebuiltWDA`: `true`
    > - `appium:useXctestrunFile`: `false`
    > - `appium:skipLogCapture`: `true`
    - `appium:wdaBaseUrl`: `http://localhost`

pencil1, 2021-02-22:
[image]
还是这样的报错，看最后这句，好像appium要自己启动wda

electricbubble, 2021-02-22:
1. `Unable to start WebDriverAgent session because of xcodebuild failure`, 原因也是 `404`, 和 `/status` 的问题一样
2. 为什么是 `404`? 我猜是 `this.getUrlForProxy(url)` 出了差错, 请求使用的是 `newUrl`, 日志输出的是 `url`
```js
  async proxy (url, method, body = null) {
    method = method.toUpperCase();
    const newUrl = this.getUrlForProxy(url);
    const truncateBody = (content) => _.truncate(
      _.isString(content) ? content : JSON.stringify(content),
      { length: MAX_LOG_BODY_LENGTH });
    const reqOpts = {
      url: newUrl,
      method,
      headers: {
        'content-type': 'application/json; charset=utf-8',
        'user-agent': 'appium',
        accept: 'application/json, */*',
      },
      proxy: false,
      timeout: this.timeout,
      httpAgent: this.httpAgent,
      httpsAgent: this.httpsAgent,
    };
    // GET methods shouldn't have any body. Most servers are OK with this, but WebDriverAgent throws 400 errors
    if (util.hasValue(body) && method !== 'GET') {
      if (typeof body !== 'object') {
        try {
          reqOpts.data = JSON.parse(body);
        } catch (e) {
          throw new Error(`Cannot interpret the request body as valid JSON: ${truncateBody(body)}`);
        }
      } else {
        reqOpts.data = body;
      }
    }

    log.debug(`Proxying [${method} ${url || '/'}] to [${method} ${newUrl}] ` +
      (reqOpts.data ? `with body: ${truncateBody(reqOpts.data)}` : 'with no body'));
    // ...
  }
```
> https://github.com/appium/appium-base-driver/blob/5521888b0fe3496ce21238cece05c6bb16244f93/lib/jsonwp-proxy/proxy.js#L139-L172

而 `getUrlForProxy` 这个函数里的 `this.base`, 我猜是导致 `404` 的原因
```js
const proxyBase = `${this.scheme}://${this.server}:${this.port}${this.base}`;
```
> https://github.com/appium/appium-base-driver/blob/5521888b0fe3496ce21238cece05c6bb16244f93/lib/jsonwp-proxy/proxy.js#L95

---

我暂时无法确定是 `Capabilities` 设置不正确, 还是其他问题...

jongyu, 2021-03-29:
使用以下配置可以解决
```json
{
  "automationName": "XCUITest",
  "platformName": "iOS",
  "platformVersion": "",
  "deviceName": "iPhone 12 Pro",
  "bundleId": "",
  "udid": "",
  "newCommandTimeout": "3600",
  "skipLogCapture": "true",
  "webDriverAgentUrl": "http://localhost:8100",
  "usePrebuiltWDA": "true",
  "useXctestrunFile": "false"
}
```

2268314959, 2021-08-03:
@pencil1 你好，请问一下Unable to start WebDriverAgent session because of xcodebuild failure，这个问题解决了吗