alibaba/tidevice #30: pip install tidevice and prompts "command not found: tidevice"

cssshuang, 2021-02-20:
Hello, after the installation, view the version "tidevice version" prompts "command not found: tidevice", is there something wrong with the configuration?
Thanks very much!

electricbubble, 2021-02-20:
Is it installed in the virtual environment?
And run it from the command line in the global environment?

> may need to provide more details about the running environment

cssshuang, 2021-02-20:
> Is it installed in the virtual environment?
> And run it from the command line in the global environment?
> 
> > may need to provide more details about the running environment

MBP ~ % pip install tidevice
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: tidevice in ./Library/Python/3.8/lib/python/site-packages (0.1.13)
Requirement already satisfied: logzero in ./Library/Python/3.8/lib/python/site-packages (from tidevice) (1.6.3)
Requirement already satisfied: tornado in ./Library/Python/3.8/lib/python/site-packages (from tidevice) (6.1)
Requirement already satisfied: simple-tornado in ./Library/Python/3.8/lib/python/site-packages (from tidevice) (0.2.0)
Requirement already satisfied: requests in ./Library/Python/3.8/lib/python/site-packages (from tidevice) (2.25.1)
Requirement already satisfied: cached-property in ./Library/Python/3.8/lib/python/site-packages (from tidevice) (1.5.2)
Requirement already satisfied: Pillow in ./Library/Python/3.8/lib/python/site-packages (from tidevice) (8.1.0)
Requirement already satisfied: packaging in ./Library/Python/3.8/lib/python/site-packages (from tidevice) (20.9)
Requirement already satisfied: colored in ./Library/Python/3.8/lib/python/site-packages (from tidevice) (1.4.2)
Requirement already satisfied: pyparsing>=2.0.2 in ./Library/Python/3.8/lib/python/site-packages (from packaging->tidevice) (2.4.7)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in ./Library/Python/3.8/lib/python/site-packages (from requests->tidevice) (1.26.3)
Requirement already satisfied: chardet<5,>=3.0.2 in ./Library/Python/3.8/lib/python/site-packages (from requests->tidevice) (4.0.0)
Requirement already satisfied: idna<3,>=2.5 in ./Library/Python/3.8/lib/python/site-packages (from requests->tidevice) (2.10)
Requirement already satisfied: certifi>=2017.4.17 in ./Library/Python/3.8/lib/python/site-packages (from requests->tidevice) (2020.12.5)
MBP ~ % tidevice version
zsh: command not found: tidevice
MBP ~ %

electricbubble, 2021-02-20:
make sure `$PATH` is valid
```bash
cat ~/.zshrc
```

> https://stackoverflow.com/questions/31615322/zsh-conda-pip-installs-command-not-found/55860128#55860128

codeskyblue, 2021-02-22:
Maybe you can try `python3 -m tidevice version`

cssshuang, 2021-02-22:
> Maybe you can try `python3 -m tidevice version`

MacBook-Pro ~ % python3 -m tidevice version
tidevice version 0.1.13

cssshuang, 2021-02-22:
> make sure `$PATH` is valid
> 
> ```shell
> cat ~/.zshrc
> ```
> 
> > https://stackoverflow.com/questions/31615322/zsh-conda-pip-installs-command-not-found/55860128#55860128

Thanks, I tried your methods and It still doesn't work.

closed 2021-02-22 by cssshuang

Fun0623, 2021-12-12:
I just tried it, the test is useful, thank you very much

Fun0623, 2021-12-12:
> Maybe you can try `python3 -m tidevice version`

I just tried it, the test is useful, thank you very much

Mamong, 2022-12-01:
you'd better check the path '~/Library/Python/' whether there exists more that one python version or not，then find which folder tidevice locates in.And add env in ~/.zshrc: export PATH="$HOME/Library/Python/3.8/bin:$PATH", then excute 'source ~/.zshrc'.