alibaba/tidevice #32: Unable to build WebDriverAgentRunner ipa file

vikram-chaitanya, 2021-02-26:
Hi,

I am trying to build web driver agent ipa, by following your DEVELOP.md instructions, but noticed that the xcode project doesn't have a scheme WebDriverAgent. Can you please provide instructions on how to build the ipa from xcode/xcodebuild.

electricbubble, 2021-02-26:
```bash
xcodebuild build-for-testing -workspace WebDriverAgent.xcworkspace/ -scheme WebDriverAgentRunner \
-sdk iphoneos -configuration Release -derivedDataPath /tmp/derivedDataPath
```

> -scheme WebDriverAgentRunner

vikram-chaitanya, 2021-02-28:
@electricbubble  @codeskyblue 

Thanks for the help. I was able to build it once changed the command accordingly. But got struck at the next to sign this app using isign(https://github.com/apperian/isign). I am able to sign every other app using isign but this one fails. I am not sure why this is happening. I am getting this error when try to sign the app. Can you please help? 

```
bash-3.2$ isign -c ~/.isign/certificate.pem -k ~/.isign/key.pem -p ~/.isign/isign.mobileprovision -o /tmp/derivedDataPath/Build/Products/Release-iphoneos/WebDriverAgentRunner-Runner-Resigned.app /tmp/derivedDataPath/Build/Products/Release-iphoneos/WebDriverAgentRunner-Runner.app
Traceback (most recent call last):
  File "/usr/local/bin/isign", line 4, in <module>
    __import__('pkg_resources').run_script('isign==1.6.15.1614128083.dev77+root', 'isign')
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 666, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 1453, in run_script
    exec(code, namespace, namespace)
  File "/Library/Python/2.7/site-packages/isign-1.6.15.1614128083.dev77+root-py2.7.egg/EGG-INFO/scripts/isign", line 199, in <module>
    isign.resign(app_path, **kwargs)
  File "/Library/Python/2.7/site-packages/isign-1.6.15.1614128083.dev77+root-py2.7.egg/isign/isign.py", line 79, in resign
    alternate_entitlements_path)
  File "/Library/Python/2.7/site-packages/isign-1.6.15.1614128083.dev77+root-py2.7.egg/isign/archive.py", line 401, in resign
    ua.bundle.resign(signer, provisioning_profile, alternate_entitlements_path)
  File "/Library/Python/2.7/site-packages/isign-1.6.15.1614128083.dev77+root-py2.7.egg/isign/bundle.py", line 259, in resign
    super(App, self).resign(signer)
  File "/Library/Python/2.7/site-packages/isign-1.6.15.1614128083.dev77+root-py2.7.egg/isign/bundle.py", line 176, in resign
    self.sign(signer)
  File "/Library/Python/2.7/site-packages/isign-1.6.15.1614128083.dev77+root-py2.7.egg/isign/bundle.py", line 143, in sign
    framework.resign(signer)
  File "/Library/Python/2.7/site-packages/isign-1.6.15.1614128083.dev77+root-py2.7.egg/isign/bundle.py", line 176, in resign
    self.sign(signer)
  File "/Library/Python/2.7/site-packages/isign-1.6.15.1614128083.dev77+root-py2.7.egg/isign/bundle.py", line 171, in sign
    executable = self.signable_class(self, self.get_executable_path(), signer)
  File "/Library/Python/2.7/site-packages/isign-1.6.15.1614128083.dev77+root-py2.7.egg/isign/signable.py", line 43, in __init__
    self.m = macho.MachoFile.parse_stream(self.f)
  File "/Library/Python/2.7/site-packages/construct-2.5.2-py2.7.egg/construct/core.py", line 198, in parse_stream
    return self._parse(stream, Container())
  File "/Library/Python/2.7/site-packages/construct-2.5.2-py2.7.egg/construct/core.py", line 670, in _parse
    subobj = sc._parse(stream, context)
  File "/Library/Python/2.7/site-packages/construct-2.5.2-py2.7.egg/construct/core.py", line 851, in _parse
    obj = self.cases.get(key, self.default)._parse(stream, context)
  File "/Library/Python/2.7/site-packages/construct-2.5.2-py2.7.egg/construct/core.py", line 670, in _parse
    subobj = sc._parse(stream, context)
  File "/Library/Python/2.7/site-packages/construct-2.5.2-py2.7.egg/construct/core.py", line 445, in _parse
    raise ArrayError("expected %d, found %d" % (count, c), sys.exc_info()[1])
construct.core.ArrayError: ('expected 4, found 3', ArrayError('expected 26, found 5', MappingError('no decoding mapping for 2147483700 [cmd]',)))

```

codeskyblue, 2021-03-02:
Try this: https://dantheman827.github.io/ios-app-signer/

vikram-chaitanya, 2021-03-15:
@electricbubble @codeskyblue 

Thanks for the suggestions. I tried to see if its possible to re-sign the web driver agent ipa file in linux system, but since it seems like its not working out, I has to settle for macOS system. If you know some way of signing this in linux please do let me know. Anyway thanks again for the suggestions.

closed 2021-03-15 by vikram-chaitanya

codeskyblue, 2021-05-08:
@vikram-chaitanya How about this tool. https://github.com/sauce-archives/isign works on Linux