alibaba/tidevice #35: 性能方面有关于电池相关的吗？比如电压，电流，耗电量这种

Huyusha25, 2021-03-05:


codeskyblue, 2021-03-06:
```python
import tidevice

d  = tidevice.Device()
print(d.get_io_power())
```

Output example

```text
{'Diagnostics': {'IORegistry': {'AbsoluteCapacity': 653,
   'AdapterDetails': {'AdapterID': 0,
    'Current': 2100,
    'Description': 'usb host',
    'FamilyCode': -536854528,
    'IsWireless': False,
    'PMUConfiguration': 2100,
    'SharedSource': 0,
    'Source': 0,
    'Voltage': 5000,
    'Watts': 11},
   'AdapterInfo': 0,
   'Amperage': 1602,
   'AppleChargeRateLimitIndex': 0,
   'AppleRawAdapterDetails': [{'AdapterID': 0,
     'Current': 2100,
     'Description': 'usb host',
     'FamilyCode': -536854528,
     'IsWireless': False,
     'PMUConfiguration': 2100,
     'SharedSource': 0,
     'Source': 0,
     'Voltage': 5000,
     'Watts': 11},
    {'AdapterID': 0,
     'Current': 0,
     'Description': 'batt',
     'ErrorFlags': 0,
     'IsWireless': False,
     'PMUConfiguration': 2100,
     'SharedSource': 0,
     'Source': 0,
     'Voltage': 5000,
     'Watts': 0}],
   'AppleRawBatteryVoltage': 3876,
   'AppleRawCurrentCapacity': 612,
   'AppleRawExternalConnected': True,
   'AppleRawMaxCapacity': 2900,
   'AtCriticalLevel': False,
   'AvgTimeToFull': 69,
   'BatteryCellDisconnectCount': 0,
   'BatteryData': {'AlgoChemID': 1008572310,
    'BatteryHealthMetric': 125,
    'CellCurrentAccumulator': [0],
    'CellCurrentAccumulatorCount': 0,
    'CellVoltage': [3876],
    'CellWom': [3159088],
    'ChargeAccum': -394,
    'ChemID': 1007911017,
    'ChemicalWeightedRa': 121,
    'CycleCount': 39,
    'DOD0': [119],
    'DailyMaxSoc': 100,
    'DailyMinSoc': 11,
    'DesignCapacity': 2798,
    'FccComp1': 2900,
    'FccComp2': 2900,
    'FilteredCurrent': 1295,
    'FilteredCurrentRc3': 1585,
    'FilteredCurrentRc4': 397,
[... 131 lines cut ...]
   'InstantAmperage': 1582,
   'IsCharging': True,
   'KioskMode': {'KioskModeFullChargeVoltage': 4374,
    'KioskModeHighSocDays': 0,
    'KioskModeHighSocSeconds': 0,
    'KioskModeLastHighSocHours': 0,
    'KioskModeMode': 1},
   'LPEMData': {'Mode': 0},
   'Location': 0,
   'ManufacturerData': b'FG904374FYEQ27C2Y\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
   'MaxCapacity': 100,
   'NominalChargeCapacity': 2874,
   'OCVData': {},
   'PostChargeWaitSeconds': 120,
   'PostDischargeWaitSeconds': 120,
   'Serial': 'FG904533579Q6893E',
   'Temperature': 2659,
   'TimeRemaining': 69,
   'UpdateTime': 1615031798,
   'UserVisiblePathUpdated': 1615031792,
   'VirtualTemperature': 2539,
   'Voltage': 3876,
   'built-in': True}},
 'Status': 'Success'}
```

codeskyblue, 2021-03-06:
耗电量，我暂时不太清楚怎么搞

yunhailu, 2022-02-08:
> ```python
> import tidevice
> 
> d  = tidevice.Device()
> print(d.get_io_power())
> ```
> 
> Output example
> 
> ```
> {'Diagnostics': {'IORegistry': {'AbsoluteCapacity': 653,
>    'AdapterDetails': {'AdapterID': 0,
>     'Current': 2100,
>     'Description': 'usb host',
>     'FamilyCode': -536854528,
>     'IsWireless': False,
>     'PMUConfiguration': 2100,
>     'SharedSource': 0,
>     'Source': 0,
>     'Voltage': 5000,
>     'Watts': 11},
>    'AdapterInfo': 0,
>    'Amperage': 1602,
>    'AppleChargeRateLimitIndex': 0,
>    'AppleRawAdapterDetails': [{'AdapterID': 0,
>      'Current': 2100,
>      'Description': 'usb host',
>      'FamilyCode': -536854528,
>      'IsWireless': False,
>      'PMUConfiguration': 2100,
>      'SharedSource': 0,
>      'Source': 0,
>      'Voltage': 5000,
>      'Watts': 11},
>     {'AdapterID': 0,
>      'Current': 0,
>      'Description': 'batt',
>      'ErrorFlags': 0,
>      'IsWireless': False,
>      'PMUConfiguration': 2100,
>      'SharedSource': 0,
>      'Source': 0,
>      'Voltage': 5000,
>      'Watts': 0}],
>    'AppleRawBatteryVoltage': 3876,
>    'AppleRawCurrentCapacity': 612,
>    'AppleRawExternalConnected': True,
>    'AppleRawMaxCapacity': 2900,
>    'AtCriticalLevel': False,
>    'AvgTimeToFull': 69,
>    'BatteryCellDisconnectCount': 0,
>    'BatteryData': {'AlgoChemID': 1008572310,
>     'BatteryHealthMetric': 125,
>     'CellCurrentAccumulator': [0],
>     'CellCurrentAccumulatorCount': 0,
>     'CellVoltage': [3876],
>     'CellWom': [3159088],
>     'ChargeAccum': -394,
>     'ChemID': 1007911017,
>     'ChemicalWeightedRa': 121,
>     'CycleCount': 39,
>     'DOD0': [119],
>     'DailyMaxSoc': 100,
>     'DailyMinSoc': 11,
>     'DesignCapacity': 2798,
>     'FccComp1': 2900,
>     'FccComp2': 2900,
>     'FilteredCurrent': 1295,
>     'FilteredCurrentRc3': 1585,
>     'FilteredCurrentRc4': 397,
[... 133 lines cut ...]
>    'KioskMode': {'KioskModeFullChargeVoltage': 4374,
>     'KioskModeHighSocDays': 0,
>     'KioskModeHighSocSeconds': 0,
>     'KioskModeLastHighSocHours': 0,
>     'KioskModeMode': 1},
>    'LPEMData': {'Mode': 0},
>    'Location': 0,
>    'ManufacturerData': b'FG904374FYEQ27C2Y\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
>    'MaxCapacity': 100,
>    'NominalChargeCapacity': 2874,
>    'OCVData': {},
>    'PostChargeWaitSeconds': 120,
>    'PostDischargeWaitSeconds': 120,
>    'Serial': 'FG904533579Q6893E',
>    'Temperature': 2659,
>    'TimeRemaining': 69,
>    'UpdateTime': 1615031798,
>    'UserVisiblePathUpdated': 1615031792,
>    'VirtualTemperature': 2539,
>    'Voltage': 3876,
>    'built-in': True}},
>  'Status': 'Success'}
> ```

BatteryHealthMetric, how to calculate this key？