alibaba/tidevice #36: 连线的时候有机率出现断线的情况, 请问一下我该如何处理?

highboy168, 2021-03-09:
我在连线的时候遇到了一些问题, 当我使用facebook-wda运行tidevice时,
有机率出现断线的情况, 请问一下我该如何处理?

- device:iphone12 mini
- iOS:14.4
- xcode:12.4

[I 210309 13:21:00 _device:887] logProcess:     t =  1907.73s         Find: Identity Binding
[I 210309 13:21:00 _device:887] logProcess:     t =  1907.74s     Check for interrupting elements affecting "inputAccount" Other
[I 210309 13:21:00 _device:887] logProcess:     t =  1907.74s         Requesting snapshot of accessibility hierarchy for app with pid 1808
[I 210309 13:21:00 _device:1059] xctrunner quited

facebook-wda:
requests.exceptions.ConnectionError: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))

codeskyblue, 2021-03-09:
排除一下数据线问题，接口问题，usbhub电压问题。
如果还是出现，tidevice wdaproxy有自动重启xcuitest的功能

codeskyblue, 2021-03-23:
今天看到了这种数据线感觉不错。
https://item.jd.com/44473991638.html

[image]