alibaba/tidevice #43: 无法创建Session，Unexpected end of file from server

jongyu, 2021-03-17:
系统：macOS Big Sur(11.2.3)
手机：iPhone 12 Pro Max(14.4.1)
创建Session时提示Unexpected end of file from server
wda显示的日志：
```
[D 210317 22:46:10 _relay:36] handle stream from: ('::1', 50265, 0, 0)
[D 210317 22:56:18 _relay:36] handle stream from: ('127.0.0.1', 50619)
[D 210317 22:56:46 _wdaproxy:91] [00008101] WDA is not response, check again after 10s
[I 210317 22:56:59 _wdaproxy:95] [00008101] WDA confirmed not running
[I 210317 22:56:59 _wdaproxy:104] [00008101] WDA keeper stopped
[I 210317 22:56:59 _wdaproxy:132] [00008101] WDA stopped for the 5st time, running 11.8 minutes
[D 210317 22:56:59 _wdaproxy:116] [00008101] launch WDA
```

jongyu, 2021-03-17:
使用的命令是`tidevice wdaproxy --port 8100`
后面我使用`tidevice xctest --debug` + `iproxy 8100 8100` 可以正常使用，不知道是不是大佬代理wda服务的问题

closed 2021-03-18 by jongyu

codeskyblue, 2021-03-23:
可能是检查的wda的时间问题太短。