alibaba/tidevice #45: 执行tidevice xctest报错为xctrunner quited

xiaoxiaoshao, 2021-03-23:
安装好wda后，在window中执行tidevice xctest报错为xctrunner quited，麻烦大佬帮忙看下
[image]

codeskyblue, 2021-03-23:
这退出的够快的啊
用下面的这个命令跑跑看看

```
DEBUG=true tidevice xctest  
```

xiaoxiaoshao, 2021-03-23:
debug了下，发现貌似没有print ServerURLHere ,我看代码注释说In low iOS versions, 'Using singleton test manager' may not be printed... mark wda launch status = True if server url has been printed，但是我的版本蛮高的14.4.1，有没有可能我wda包有问题~

[image]

codeskyblue, 2021-03-24:
你好像装错包了

xiaoxiaoshao, 2021-03-24:
> debug了下，发现貌似没有print ServerURLHere ,我看代码注释说In low iOS versions, 'Using singleton test manager' may not be printed... mark wda launch status = True if server url has been printed，但是我的版本蛮高的14.4.1，有没有可能我wda包有问题~
> 
> [image]

现在可以正常启动了，果然装的包有问题，xcode版本太高，在网上找攻略修复了下，谢谢大佬

closed 2021-03-24 by xiaoxiaoshao

Zekylly, 2021-04-14:
> > debug了下，发现貌似没有print ServerURLHere ,我看代码注释说In low iOS versions, 'Using singleton test manager' may not be printed... mark wda launch status = True if server url has been printed，但是我的版本蛮高的14.4.1，有没有可能我wda包有问题~
> > [image]
> 
> 现在可以正常启动了，果然装的包有问题，xcode版本太高，在网上找攻略修复了下，谢谢大佬

我换了低版本的xcode11.6，还是有这个问题，不知道怎么解决
zekylldeMacBook-Pro:Project zekyll$ tidevice xctest
[I 210414 13:35:20 _device:990] BundleID: com.facebook.IntegrationTestszek.xctrunner
[I 210414 13:35:20 _device:992] DeviceIdentifier: 5975536dd9ce12889a2873ce82ecb302f8347a9f
[I 210414 13:35:20 _device:834] SignIdentity: 'Apple Development: saxono@163.com (YU39XU98SG)'
[I 210414 13:35:20 _device:840] CFBundleExecutable: IntegrationTests_3-Runner
[I 210414 13:35:20 _device:871] AppContainer: /private/var/mobile/Containers/Data/Application/643BE368-247F-4E53-B1C8-5A221FA0F27B
[I 210414 13:35:20 _device:911] Launch 'com.facebook.IntegrationTestszek.xctrunner' pid: 893
[I 210414 13:35:20 _device:1086] ProductVersion: 12.2
[I 210414 13:35:20 _device:1116] xctrunner quited

codeskyblue, 2021-04-19:
CFBundleExecutable: IntegrationTests_3-Runner 这个就不太正常啊
应该是WebDriverAgent-Runner才对

FiveMiles, 2021-05-04:
> 安装好wda后，在window中执行tidevice xctest报错为xctrunner quited，麻烦大佬帮忙看下
> [image]

大佬解决了嘛？

Abcefg123, 2021-05-20:
> > debug了下，发现貌似没有print ServerURLHere ,我看代码注释说In low iOS versions, 'Using singleton test manager' may not be printed... mark wda launch status = True if server url has been printed，但是我的版本蛮高的14.4.1，有没有可能我wda包有问题~
> > [image]
> 
> 现在可以正常启动了，果然装的包有问题，xcode版本太高，在网上找攻略修复了下，谢谢大佬

请问咋修复的，是换个低版本的xcode重新编译一次facebook的那个WDA吗？

jeniva, 2022-03-16:
> 我这边手机有wda但是ios手机连不上，我输入下面的命令提示[xctrunner quited]
> ```
> DEBUG=true tidevice wdaproxy 
> ```

[D 220316 11:09:32 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0490B930> closed
[D 220316 11:09:32 __main__:72] AutoComplete udid 00008020-000D78103499002E
[D 220316 11:09:32 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0490B930> closed
[I 220316 11:09:32 _wdaproxy:125] [00008020-000D78103499002E] WDA check every 30.0 seconds
[D 220316 11:09:32 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0490BFD0> closed
[D 220316 11:09:32 _wdaproxy:131] [00008020-000D78103499002E] launch WDA
[D 220316 11:09:32 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0490BAB0> closed
[D 220316 11:09:32 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 42015, 'ProgName': 'tidevice'}
[D 220316 11:09:32 _device:283] connect port: 42015
[D 220316 11:09:32 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0491F370> closed
[D 220316 11:09:32 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 42015, 'ProgName': 'tidevice'}
[D 220316 11:09:32 _device:283] connect port: 42015
[D 220316 11:09:37 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x05789BB0> closed
[D 220316 11:09:37 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x05789B50> closed
Traceback (most recent call last):
  File "manis-tidevice.py", line 7, in <module>
  File "tidevice\__main__.py", line 809, in main
  File "tidevice\__main__.py", line 399, in cmd_relay
  File "tidevice\_relay.py", line 84, in relay
  File "tornado\tcpserver.py", line 151, in listen
  File "tornado\netutil.py", line 161, in bind_sockets
OSError: [WinError 10048] ͨ▒▒ÿ▒▒▒׽▒▒ֵ▒ַ(Э▒▒/▒▒▒▒▒ַ/▒˿▒)ֻ▒▒▒▒ʹ▒▒һ▒Ρ▒
[21100] Failed to execute script 'manis-tidevice' due to unhandled exception!
[D 220316 11:09:37 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x04839D10> closed
[D 220316 11:09:37 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x04839CB0> closed
[D 220316 11:09:37 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:09:37 _device:283] connect port: 32498
[D 220316 11:09:37 _safe_socket:149] Recv pair record data ...
[D 220316 11:09:37 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0484B370> closed
[D 220316 11:09:38 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0484B350> closed
[D 220316 11:09:38 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 33527, 'ProgName': 'tidevice'}
[D 220316 11:09:38 _device:283] connect port: 33527
[I 220316 11:09:38 _device:958] BundleID: com.ezviz.WebDriverAgentRunner.xctrunner
[D 220316 11:09:38 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:09:38 _device:283] connect port: 32498
[D 220316 11:09:38 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x04839EF0> closed
[I 220316 11:09:38 _device:961] ProductVersion: 14.6
[I 220316 11:09:38 _device:963] DeviceIdentifier: 00008020-000D78103499002E
[D 220316 11:09:38 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:09:38 _device:283] connect port: 32498
[D 220316 11:09:38 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0484BE10> closed
[D 220316 11:09:38 _device:929] ProductVersion: 14.6
[D 220316 11:09:38 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:09:38 _device:283] connect port: 32498
[D 220316 11:09:38 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0484B810> closed
[D 220316 11:09:38 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 34807, 'ProgName': 'tidevice'}
[D 220316 11:09:38 _device:283] connect port: 34807
[D 220316 11:09:38 _instruments:591] DTXMessage: expects_reply:0 flags:2 ('_notifyOfPublishedCapabilities:', [{'com.apple.private.DTXBlockCompression': 2, 'com.apple.private.DTXConnection': 1}])
[D 220316 11:09:38 _instruments:591] DTXMessage: expects_reply:0 flags:0 None
[D 220316 11:09:38 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:09:38 _device:283] connect port: 32498
[D 220316 11:09:39 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x048579B0> closed
[D 220316 11:09:39 _device:929] ProductVersion: 14.6
[D 220316 11:09:39 _instruments:591] DTXMessage: expects_reply:0 flags:3 36
[D 220316 11:09:39 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:09:39 _device:283] connect port: 32498
[D 220316 11:09:39 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0486F8B0> closed
[D 220316 11:09:39 _device:929] ProductVersion: 14.6
[D 220316 11:09:39 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:09:39 _device:283] connect port: 32498
[D 220316 11:09:39 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0486FB50> closed
[... cut ...]
[D 220316 11:09:41 _instruments:591] DTXMessage: expects_reply:1 flags:2 ('_XCT_logDebugMessage:', ['11:09:42.546 WebDriverAgentRunner-Runner[896:440555] Returning result XCTWaiterResultCompleted for waiter <XCTWaiter: 0x283b6b170, state: XCTWaiterStateFinished>\n'])
[D 220316 11:09:45 _instruments:569] dtxm socket closed
[I 220316 11:09:45 _device:1084] xctrunner quited
[D 220316 11:09:45 _instruments:591] DTXMessage: expects_reply:0 flags:2 ('pidDiedCallback:', [896])
[D 220316 11:09:45 _instruments:606] Ignore notification from server: 24, 0x2, ('pidDiedCallback:', [896])

jeniva, 2022-03-16:
$ DEBUG=true tidevice.exe xctest -B com.xxx.WebDriverAgentRunner.xctrunner
[D 220316 11:13:59 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0473B910> closed
[D 220316 11:13:59 __main__:72] AutoComplete udid 00008020-000D78103499002E
[D 220316 11:13:59 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0473B8B0> closed
[D 220316 11:13:59 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:13:59 _device:283] connect port: 32498
[D 220316 11:13:59 _safe_socket:149] Recv pair record data ...
[D 220316 11:13:59 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0473B910> closed
[D 220316 11:13:59 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0473BF30> closed
[D 220316 11:13:59 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 22776, 'ProgName': 'tidevice'}
[D 220316 11:13:59 _device:283] connect port: 22776
[I 220316 11:13:59 _device:958] BundleID: com.ezviz.WebDriverAgentRunner.xctrunner
[D 220316 11:13:59 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:13:59 _device:283] connect port: 32498
[D 220316 11:13:59 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0473BF10> closed
[I 220316 11:13:59 _device:961] ProductVersion: 14.6
[I 220316 11:13:59 _device:963] DeviceIdentifier: 00008020-000D78103499002E
[D 220316 11:13:59 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:13:59 _device:283] connect port: 32498
[D 220316 11:13:59 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0475E790> closed
[D 220316 11:13:59 _device:929] ProductVersion: 14.6
[D 220316 11:13:59 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:13:59 _device:283] connect port: 32498
[D 220316 11:14:00 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0475EF70> closed
[D 220316 11:14:00 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 24312, 'ProgName': 'tidevice'}
[D 220316 11:14:00 _device:283] connect port: 24312
[D 220316 11:14:00 _instruments:591] DTXMessage: expects_reply:0 flags:2 ('_notifyOfPublishedCapabilities:', [{'com.apple.private.DTXBlockCompression': 2, 'com.apple.private.DTXConnection': 1}])
[D 220316 11:14:00 _instruments:591] DTXMessage: expects_reply:0 flags:0 None
[D 220316 11:14:00 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:14:00 _device:283] connect port: 32498
[D 220316 11:14:00 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0475E790> closed
[D 220316 11:14:00 _device:929] ProductVersion: 14.6
[D 220316 11:14:00 _instruments:591] DTXMessage: expects_reply:0 flags:3 36
[D 220316 11:14:00 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:14:00 _device:283] connect port: 32498
[D 220316 11:14:00 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0474C450> closed
[D 220316 11:14:00 _device:929] ProductVersion: 14.6
[D 220316 11:14:00 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:14:00 _device:283] connect port: 32498
[D 220316 11:14:00 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0472BCF0> closed
[D 220316 11:14:00 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 25592, 'ProgName': 'tidevice'}
[D 220316 11:14:00 _device:283] connect port: 25592
[D 220316 11:14:00 _instruments:591] DTXMessage: expects_reply:0 flags:2 ('_notifyOfPublishedCapabilities:', [{'com.apple.private.DTXBlockCompression': 2, 'com.apple.private.DTXConnection': 1}])
[D 220316 11:14:00 _instruments:591] DTXMessage: expects_reply:0 flags:0 None
[D 220316 11:14:00 _instruments:591] DTXMessage: expects_reply:0 flags:3 36
[D 220316 11:14:00 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:14:00 _device:283] connect port: 32498
[D 220316 11:14:00 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x0477C4B0> closed
[D 220316 11:14:00 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 26360, 'ProgName': 'tidevice'}
[D 220316 11:14:00 _device:283] connect port: 26360
[I 220316 11:14:00 _device:802] SignIdentity: 'Apple Development: Molly Fukuchi (35PHN77S4R)'
[I 220316 11:14:00 _device:808] CFBundleExecutable: WebDriverAgentRunner-Runner
[D 220316 11:14:00 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 32498, 'ProgName': 'tidevice'}
[D 220316 11:14:00 _device:283] connect port: 32498
[D 220316 11:14:00 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x04767DD0> closed
[D 220316 11:14:00 _device:281] Send payload: {'DeviceID': 9, 'MessageType': 'Connect', 'PortNumber': 27128, 'ProgName': 'tidevice'}
[D 220316 11:14:00 _device:283] connect port: 27128
[D 220316 11:14:00 _device:822] remove /tmp/WebDriverAgentRunner-1D62DA06-2890-419A-8B56-826E3582671C.xctestconfiguration
[... cut ...]
[D 220316 14:25:01 _instruments:569] dtxm socket closed
[D 220316 14:25:01 _instruments:591] DTXMessage: expects_reply:0 flags:2 ('pidDiedCallback:', [338])
[D 220316 14:25:01 _instruments:606] Ignore notification from server: 9, 0x2, ('pidDiedCallback:', [338])
[D 220316 14:25:01 _safe_socket:82] Socket <tidevice._safe_socket.PlistSocket object at 0x04C4C8B0> closed
[D 220316 14:25:01 _device:929] ProductVersion: 14.6
[D 220316 14:25:01 _instruments:591] DTXMessage: expects_reply:0 flags:3 True
[I 220316 14:25:01 _device:1084] xctrunner quited