alibaba/tidevice #49: xctrunner quited退出问题

zhouyuxia0204, 2021-03-24:
[D 210324 18:52:55 _device:942] logProcess: t = 24.55s Requesting snapshot of accessibility hierarchy for app with pid 3966
Exception in thread Thread-2:
Traceback (most recent call last):
File "d:\appstore\python37\lib\threading.py", line 926, in _bootstrap_inner
self.run()
File "d:\appstore\python37\lib\threading.py", line 870, in run
self._target(*self._args, **self._kwargs)
File "d:\appstore\python37\lib\site-packages\tidevice_instruments.py", line 558, in _drain
self._drain_single_message()
File "d:\appstore\python37\lib\site-packages\tidevice_instruments.py", line 578, in _drain_single_message
mheader, payload = self.recv_dtx_message()
File "d:\appstore\python37\lib\site-packages\tidevice_instruments.py", line 499, in recv_dtx_message
assert mheader.conversation_index in [0, 1, 2]
AttributeError: 'NoneType' object has no attribute 'conversation_index'
[I 210324 18:52:55 _device:1114]
xctrunner quited

什么原因造成的运行中途退出，随机出现此问题

name-jiang, 2021-04-23:
我运行也遇到了该类问题，不知道咋解决

name-jiang, 2021-04-23:
不知道问题是不是一样的，我运行也是报这个错误ft10_1: Traceback (most recent call last):
  File "D:\python3\lib\site-packages\urllib3\connectionpool.py", line 677, in urlopen
    chunked=chunked,
  File "D:\python3\lib\site-packages\urllib3\connectionpool.py", line 392, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "D:\python3\lib\http\client.py", line 1277, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "D:\python3\lib\http\client.py", line 1323, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "D:\python3\lib\http\client.py", line 1272, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "D:\python3\lib\http\client.py", line 1032, in _send_output
    self.send(msg)
  File "D:\python3\lib\http\client.py", line 972, in send
    self.connect()
  File "D:\python3\lib\site-packages\wda\requests_usbmux.py", line 54, in connect
    conn = _device.create_inner_connection(int(port))
  File "D:\python3\lib\site-packages\wda\usbmux.py", line 239, in create_inner_connection
    "device port:{} is not ready".format(_original_port))
wda.exceptions.MuxConnectError: device port:8100 is not ready

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\python3\lib\site-packages\requests\adapters.py", line 449, in send
    timeout=timeout
  File "D:\python3\lib\site-packages\urllib3\connectionpool.py", line 727, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "D:\python3\lib\site-packages\urllib3\util\retry.py", line 410, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "D:\python3\lib\site-packages\urllib3\packages\six.py", line 734, in reraise
    raise value.with_traceback(tb)
  File "D:\python3\lib\site-packages\urllib3\connectionpool.py", line 677, in urlopen
    chunked=chunked,
  File "D:\python3\lib\site-packages\urllib3\connectionpool.py", line 392, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "D:\python3\lib\http\client.py", line 1277, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "D:\python3\lib\http\client.py", line 1323, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "D:\python3\lib\http\client.py", line 1272, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "D:\python3\lib\http\client.py", line 1032, in _send_output
    self.send(msg)
  File "D:\python3\lib\http\client.py", line 972, in send
    self.connect()
  File "D:\python3\lib\site-packages\wda\requests_usbmux.py", line 54, in connect
    conn = _device.create_inner_connection(int(port))
  File "D:\python3\lib\site-packages\wda\usbmux.py", line 239, in create_inner_connection
    "device port:{} is not ready".format(_original_port))
urllib3.exceptions.ProtocolError: ('Connection aborted.', MuxConnectError('device port:8100 is not ready'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\app_test_ios\scripts\sgapp_home_userdefine_click.py", line 25, in setUp
    cold_startapp(client)
  File "D:\app_test_ios\base\sg_action.py", line 46, in cold_startapp
    if client.locked():  #判断是否锁屏
  File "D:\python3\lib\site-packages\wda\__init__.py", line 496, in locked
    return self.http.get("/wda/locked").value
  File "D:\python3\lib\site-packages\wda\utils.py", line 47, in _inner
    return fn(*args, **kwargs)
  File "D:\python3\lib\site-packages\wda\__init__.py", line 454, in _fetch
    response = httpdo(url, method, data, timeout)
  File "D:\python3\lib\site-packages\wda\__init__.py", line 124, in httpdo
    return _unsafe_httpdo(url, method, data, timeout)
  File "D:\python3\lib\site-packages\wda\__init__.py", line 154, in _unsafe_httpdo
    timeout=timeout)
  File "D:\python3\lib\site-packages\requests\sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "D:\python3\lib\site-packages\requests\sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "D:\python3\lib\site-packages\requests\adapters.py", line 498, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', MuxConnectError('device port:8100 is not ready'))