alibaba/tidevice #50: tidevice\_usbmux.py的device_udid_list()方法运行出现ConnectionRefusedError: [WinError 10061] 由于目标计算机积极拒绝，无法连接

linmscat, 2021-03-25:
Exception in thread DeviceRefreshThread_ios:
Traceback (most recent call last):
  File "device\device_pool.py", line 141, in refresh_devices
  File "device\ios\ios_device_pool.py", line 19, in refresh_devices0
  File "idb\idb.py", line 14, in device_udid_list
  File "tidevice\_usbmux.py", line 77, in device_udid_list
  File "tidevice\_usbmux.py", line 72, in device_list
  File "tidevice\_usbmux.py", line 45, in send_recv
  File "tidevice\_usbmux.py", line 42, in create_connection
  File "tidevice\_safe_socket.py", line 92, in __init__
  File "tidevice\_safe_socket.py", line 48, in __init__
ConnectionRefusedError: [WinError 10061] 由于目标计算机积极拒绝，无法连接。
@zy02636 @codeskyblue @alibaba-oss @373743261 @pengchenglin

codeskyblue, 2021-03-25:
Windows电脑的话，你先把iTunes装上去

linmscat, 2021-03-25:
> Windows电脑的话，你先把iTunes装上去

是用Windows电脑，不过想知道为什么需要装iTunes，这里面的原因是什么？

codeskyblue, 2021-03-25:
iTunes里面有usbmux驱动

codeskyblue, 2021-03-25:
不装iTunes也可以，装个什么iTools之类的手机助手也可以

linmscat, 2021-03-25:
> iTunes里面有usbmux驱动

这个驱动文件可以直接在脚本里面运行，从而绕过iTunes的安装吗？

linmscat, 2021-03-25:
> 不装iTunes也可以，装个什么iTools之类的手机助手也可以

这个平台有考虑优化吗，不需要装这些手机助手就能装usb需要的驱动文件？ @codeskyblue

electricbubble, 2021-03-25:
可以阅读 [wiki usbmux](https://www.theiphonewiki.com/wiki/Usbmux)

你可以选择使用 [libimobiledevice/usbmuxd](https://github.com/libimobiledevice/usbmuxd) 从而跳过 `iTunes` 或其他第三方软件

linmscat, 2021-03-25:
> libimobiledevice/usbmuxd

了解了，谢谢！