alibaba/tidevice #52: tidevice list可用但是tideivce applist 报错，之前是可以的，现在不知道为啥不行了

jeniva, 2021-03-29:
File "d:\python3.7.8\lib\site-packages\tidevice\_usbmux.py", line 81, in _check
    raise MuxReplyError(data['Number'])
  File "d:\python3.7.8\lib\site-packages\tidevice\exceptions.py", line 14, in __init__
    self.reply_code = UsbmuxReplyCode(number)
  File "d:\python3.7.8\lib\enum.py", line 315, in __call__
    return cls.__new__(cls, value)
  File "d:\python3.7.8\lib\enum.py", line 569, in __new__
    raise exc
  File "d:\python3.7.8\lib\enum.py", line 553, in __new__
    result = cls._missing_(value)
  File "d:\python3.7.8\lib\enum.py", line 582, in _missing_
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 22 is not a valid UsbmuxReplyCode

[image]

jeniva, 2021-03-29:
发现可以是用的iphone7，现在不行的是iphone xs 和iphoneXR，不知道为啥了，求解？

codeskyblue, 2021-03-30:
理论上错误码 也就 1-6  参考: https://github.com/libimobiledevice/usbmuxd/blob/master/src/usbmuxd-proto.h
你这个直接跳到了22

你电脑上装的是什么助手啊？

jeniva, 2021-03-30:
电脑上装了iTunes，我发现必现的现象了：就是手机插上Windows电脑每次都提示要信任的就会出现这个错误，我的iphone7在我电脑上插上不会再提示信任的是可以用的，然后这个报错的手机到同事的电脑上不会提示也可以用。不知道每次插拔都弹信任这个弹窗是怎么控制的，如果可以一直信任应该就好了

electricbubble, 2021-03-30:
我记得我在删除 `PairRecord` 的时候有遇到 `Code 22` 的报错
重新 `Pair` 后就可以触发完成 `信任`

> 晚上有空我尝试复现

electricbubble, 2021-03-30:
> `Code 22` 是 `Pair` 时的数据错误

@jeniva  你可能使用的是 `v0.3.3` 之前的版本，升级应该可以解决

jeniva, 2021-04-01:
现在最新的版本不是version 0.1.7吗？

electricbubble, 2021-04-01:
https://github.com/alibaba/taobao-iphone-device/releases/tag/0.3.1

jeniva, 2021-04-22:
应该是手机连上电脑后没有识别成功，我更新了iTunes或者iTools可以了

pansionpan, 2023-06-29:
我也遇到了同样的问题，所以这个问题的解决办法是更新ITunes吗？我的电脑里没有装载iTools