alibaba/tidevice #53: tidevice 执行自动化的时候可以支持iPhone手机录屏功能吗

linmscat, 2021-03-30:
需要在执行自动化过程中能够录屏手机的操作过程 @zy02636 @codeskyblue @alibaba-oss @373743261 @pengchenglin

codeskyblue, 2021-03-30:
看Testerhome上有个帖子，实现了一套录屏的方案
链接：https://github.com/YueChen-C/ios-screen-record

linmscat, 2021-03-30:
> 看Testerhome上有个帖子，实现了一套录屏的方案
> 链接：https://github.com/YueChen-C/ios-screen-record

看这个环境是要在Mac电脑上，但是我用你们的框架，是Windows接着iPhone，能支持吗

linmscat, 2021-03-30:
@codeskyblue @zy02636 @alibaba-oss @373743261

codeskyblue, 2021-03-31:
我写了一个例子，使用imageio结合WDA的mjpeg server录的屏幕，你可以参考一下
https://github.com/alibaba/taobao-iphone-device/blob/main/scripts/uitest_screenrecord.py

linmscat, 2021-04-01:
> 我写了一个例子，使用imageio结合WDA的mjpeg server录的屏幕，你可以参考一下（因为是通过图片合成的视频，理论上是跨平台的
> https://github.com/alibaba/taobao-iphone-device/blob/d6071b6236976dcfd469a5175b019a9aeef209f6/scripts/uitest_screenrecord.py
好的，我试下，3q

linmscat, 2021-04-01:
> 我写了一个例子，使用imageio结合WDA的mjpeg server录的屏幕，你可以参考一下（因为是通过图片合成的视频，理论上是跨平台的
> https://github.com/alibaba/taobao-iphone-device/blob/d6071b6236976dcfd469a5175b019a9aeef209f6/scripts/uitest_screenrecord.py

运行会提示错误：
Connected to pydev debugger (build 203.7717.81)
Traceback (most recent call last):
  File "D:\Users\yl1150\AppData\Local\Programs\Python\Python38\lib\contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "F:/GitLab/taobao-iphone-device-main/scripts/uitest_screenrecord.py", line 64, in make_screenrecord
    _old_fps = c.appium_settings()['mjpegServerFramerate']
AttributeError: 'function' object has no attribute 'appium_settings'
@codeskyblue

linmscat, 2021-04-01:
> > 我写了一个例子，使用imageio结合WDA的mjpeg server录的屏幕，你可以参考一下（因为是通过图片合成的视频，理论上是跨平台的
> > https://github.com/alibaba/taobao-iphone-device/blob/d6071b6236976dcfd469a5175b019a9aeef209f6/scripts/uitest_screenrecord.py
> > 好的，我试下，3q

执行到这句：
# Read image from WDA mjpeg server
    pconn = t.create_inner_connection(9100) # default WDA mjpeg server port
    sock = pconn.get_socket()
    buf = SocketBuffer(sock)
会提示：
sock = pconn.get_socket()
AttributeError: 'PlistSocket' object has no attribute 'get_socket'
请问怎么解决 @codeskyblue

codeskyblue, 2021-04-01:
升级一下tidevice

linmscat, 2021-04-01:
> 升级一下tidevice

嗯嗯，可以了，感谢大神！

linmscat, 2021-04-01:
> 升级一下tidevice

再请假一个问题：这里为什么需要开启两个线程事件：
    stop_event = threading.Event()
    done_event = threading.Event()
    threading.Thread(target=_drain, args=(stop_event, done_event), daemon=True).start()
    yield
    stop_event.set()
    done_event.wait()
    wr.close()
@codeskyblue

linmscat, 2021-04-01:
能帮忙抽出两个方法，可以手动开始录制和结束录制的方法吗？ @codeskyblue

codeskyblue, 2021-04-01:
自己动手嘛

linmscat, 2021-04-01:
> 自己动手嘛

ok,已解决！

yu937861, 2021-04-02:
> 我写了一个例子，使用imageio结合WDA的mjpeg server录的屏幕，你可以参考一下（因为是通过图片合成的视频，理论上是跨平台的
> https://github.com/alibaba/taobao-iphone-device/blob/d6071b6236976dcfd469a5175b019a9aeef209f6/scripts/uitest_screenrecord.py

请问这个怎么用呢，执行 ：py.test -v scripts/uitest_screenrecord.py -s 后就卡在命令行了，是要自己写什么方法吗

linmscat, 2021-04-19:
这个好像不支持录制多个视频，因为WDA mjpeg server目前是写死的9100，多个视频设运行时，后面执行的时候会抢占该端口的数据，这个要怎么解决 @codeskyblue

codeskyblue, 2021-04-19:
@linmscat 你要不录完视频之后再切分

linmscat, 2021-04-19:
> @linmscat 你要不录完视频之后再切分

这个方式有点不合理，会将不同设备的录屏都耦合在一起，设备一多起来，难以管理 @codeskyblue

codeskyblue, 2021-04-19:
不同设备？你是一个设备录制一个视频的意思吗？ WDA监听的是手机内的9100端口，多个手机不冲突啊

linmscat, 2021-04-19:
对的，多个手机同时录屏

linmscat, 2021-04-19:
按这样说，这里的9100是手机上的端口，多个手机设备并不会产生冲突？

codeskyblue, 2021-04-19:
对了，包括8100也是，每个设备自己设备内监听，不冲突

linmscat, 2021-04-19:
> 对了，包括8100也是，每个设备自己设备内监

linmscat, 2021-04-19:
> 对了，包括8100也是，每个设备自己设备内监听，不冲突

Exception in thread Thread-31:
Traceback (most recent call last):
  File "D:\Users\yl1150\AppData\Local\Programs\Python\Python38\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "D:\Users\yl1150\AppData\Local\Programs\Python\Python38\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "F:\GitLab\apprunner\apprunner\utils\wda_record_helper.py", line 65, in _drain
    self._writer.append_data(im)
  File "F:\GitLab\apprunner\venv\lib\site-packages\imageio\core\format.py", line 502, in append_data
    return self._append_data(im, total_meta)
  File "F:\GitLab\apprunner\venv\lib\site-packages\imageio\plugins\ffmpeg.py", line 574, in _append_data
    self._write_gen.send(im)
ValueError: generator already executing
[swscaler @ 0000018d7fc40f80] Warning: data is not aligned! This can lead to a speed loss
@codeskyblue 这个报错是因为什么呢，后一个设备启动起来开始录屏，会报这个错误

linmscat, 2021-04-19:
执行这一步：
@pytest.fixture
def c() -> wda.Client:
    _c = wda.USBClient()
    _c.unlock()
    yield _c
会报错：   
 self._wda_client = wda.USBClient()
  File "F:\GitLab\apprunner\venv\lib\site-packages\wda\__init__.py", line 1768, in __init__
    raise RuntimeError("more then one device connected")
RuntimeError: more then one device connected
多设备貌似这一步不支持，看了下源码：
class USBClient(Client):
    """ connect device through unix:/var/run/usbmuxd """

    def __init__(self, udid: str = "", port: int = 8100, wda_bundle_id=None):
        if not udid:
            usbmux = Usbmux()
            infos = usbmux.device_list()
            if len(infos) == 0:
                raise RuntimeError("no device connected")
            elif len(infos) >= 2:
                raise RuntimeError("more then one device connected")
            udid = infos[0]['SerialNumber']

        super().__init__(url=requests_usbmux.DEFAULT_SCHEME + "{}:{}".format(udid, port))
        if self.is_ready():
            return

        _start_wda_xctest(udid, wda_bundle_id)
        if not self.wait_ready(timeout=20):
            raise RuntimeError("wda xctest launched but check failed")
     这里不支持连接两个以上的手机吗，这个是怎么回事，还有我不小心把这个issus给close了，你那边能重新开启吗？ @codeskyblue

codeskyblue, 2021-04-19:
c = wda.USBClient(设备的udid）

linmscat, 2021-04-19:
> c = wda.USBClient(设备的udid）

ok，多设备的录屏已解决，原来是没传这个udid，且调用方式也有点问题，现在可以了

suzhenyu006, 2021-11-19:
大神你好。create_inner_connection这个函数传入的本来应该默认是9100，但是我设备接入stf了，所以9100已经被转发到本地了，那这个里面的端口应该填什么呢？因为我看tidevice是自己通过uuid获取的ip地址…

suzhenyu006, 2021-11-19:
@codeskyblue 大神你好。create_inner_connection这个函数传入的本来应该默认是9100，但是我设备接入stf了，所以9100已经被转发到本地了，那这个里面的端口应该填什么呢？因为我看tidevice是自己通过uuid获取的ip地址…

gjs1990717, 2023-02-25:
/Users/gaojs/.pyenv/versions/3.10.1/bin/python3.10 "/Applications/PyCharm CE.app/Contents/plugins/python-ce/helpers/pycharm/_jb_pytest_runner.py" --target record_iOS.py::test_main
Testing started at 11:19 ...
Launching pytest with arguments record_iOS.py::test_main --no-header --no-summary -q in /Users/gaojs/PycharmProjects/airtest-dcck-performance/util

============================= test session starts ==============================
collecting ... collected 1 item

record_iOS.py::test_main FAILED                                          [100%]screenrecord is ready to begin

record_iOS.py:107 (test_main)
c = <wda.USBClient object at 0x10d2aa890>

    def test_main(c: wda.Client):
        t = tidevice.Device('00008030-0016685A3AA3C02E')
>       with make_screenrecord(c, t, "ggg.mp4"):

record_iOS.py:110: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../../../.pyenv/versions/3.10.1/lib/python3.10/contextlib.py:135: in __enter__
    return next(self.gen)
record_iOS.py:77: in make_screenrecord
    wr = imageio.get_writer(output_video_path, fps=_fps)
../../../.pyenv/versions/3.10.1/lib/python3.10/site-packages/imageio/v2.py:189: in get_writer
    image_file = imopen(uri, "w" + mode, **imopen_args)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

uri = 'ggg.mp4', io_mode = 'w?', plugin = None, extension = None
format_hint = None, legacy_mode = True, kwargs = {}
request = <imageio.core.request.Request object at 0x10d277c10>
source = 'ggg.mp4'

    def imopen(
        uri,
        io_mode,
        *,
        plugin=None,
        extension=None,
        format_hint=None,
        legacy_mode=False,
        **kwargs,
    ):
        """Open an ImageResource.
    
        .. warning::
            This warning is for pypy users. If you are not using a context manager,
            remember to deconstruct the returned plugin to avoid leaking the file
            handle to an unclosed file.
    
        Parameters
        ----------
        uri : str or pathlib.Path or bytes or file or Request
            The :doc:`ImageResource <../../user_guide/requests>` to load the
            image from.
        io_mode : str
            The mode in which the file is opened. Possible values are::
    
                ``r`` - open the file for reading
                ``w`` - open the file for writing
    
            Depreciated since v2.9:
            A second character can be added to give the reader a hint on what
            the user expects. This will be ignored by new plugins and will
            only have an effect on legacy plugins. Possible values are::
    
                ``i`` for a single image,
                ``I`` for multiple images,
                ``v`` for a single volume,
                ``V`` for multiple volumes,
                ``?`` for don't care (default)
    
[... 230 lines cut ...]
                    [
                        (
                            f"  {config.name}:  "
                            f"pip install imageio[{config.install_name}]"
                        )
                        for config in missing_plugins
                    ]
                )
                err_msg += (
                    "\nBased on the extension, the following plugins might add capable backends:\n"
                    f"{install_candidates}"
                )
    
        request.finish()
>       raise err_type(err_msg)
E       ValueError: Could not find a backend to open `ggg.mp4`` with iomode `w?`.
E       Based on the extension, the following plugins might add capable backends:
E         FFMPEG:  pip install imageio[ffmpeg]
E         pyav:  pip install imageio[pyav]

../../../.pyenv/versions/3.10.1/lib/python3.10/site-packages/imageio/core/imopen.py:298: ValueError

======================== 1 failed, 4 warnings in 0.67s =========================

Process finished with exit code 1

986379041, 2023-09-14:
> 我写了一个例子，使用imageio结合WDA的mjpeg服务器录屏，你可以参考一下（因为是通过图片合成的视频，理论上是跨平台的 [https://github.com/alibaba/taobao-iphone-设备/blob/d6071b6236976dcfd469a5175b019a9aeef209f6/scripts/uitest_screenrecord.py](https://github.com/alibaba/taobao-iphone-device/blob/d6071b6236976dcfd469a5175b019a9aeef209f6/scripts/uitest_screenrecord.py)

hello，我在使用时发现，生成的录屏视频是变速的，原本录屏时花了40s，但实际录屏文件只有15s，请问有什么办法可以解决吗？感谢！！