alibaba/tidevice #60: 求助-如何获取iOS设备总的物理内存大小？

linmscat, 2021-04-02:
需要获取设备总的内存大小，不是应用的内存大小 @codeskyblue @alibaba-oss

codeskyblue, 2021-04-02:
```
$ tidevice info --domain com.apple.disk_usage.factory --json
{
    "AmountDataAvailable": 158594158592,
    "AmountDataReserved": 209715200,
    "AmountRestoreAvailable": 165538361344,
    "CalculateDiskUsage": "OkilyDokily",
    "CalendarUsage": 4915200,
    "CameraUsage": 5071852887,
    "MediaCacheUsage": 28672,
    "NotesUsage": 0,
    "PhotoUsage": 5071852887,
    "TotalDataAvailable": 159130980352,
    "TotalDataCapacity": 249231912960,
    "TotalDiskCapacity": 256000000000,
    "TotalSystemAvailable": 0,
    "TotalSystemCapacity": 6734487552,
    "VoicemailUsage": 0,
    "WebAppCacheUsage": 0,
    "NANDInfo": "....."
}

The command is same as `ideviceinfo -q com.apple.disk_usage.factory`
As you can see the output `"TotalDiskCapacity": 256000000000,` is the capacity, 256G, it seems the unit is not Bytes, that's wired.

linmscat, 2021-04-02:
> ```
> $ tidevice info --domain com.apple.disk_usage.factory --json
> {
>     "AmountDataAvailable": 158594158592,
>     "AmountDataReserved": 209715200,
>     "AmountRestoreAvailable": 165538361344, [...]

这个获取的是手机的硬盘存储吧，我是想获得手机的运存总大小，当应用运行时占用的运存大小，目前应用运行的运存大小可以通过instrument获得，但是获取不到这台iOS手机的运存总大小？ @codeskyblue

codeskyblue, 2021-04-02:
你说的是RAM size，目前看来好像也没有能用的接口
`tidevice info --json`倒是有个看起来相关的信息

```json
{
  "NonVolatileRAM": {
        "auto-boot": "dHJ1ZQ==",
        "backlight-level": "MTUxNQ==",
        "backlight-nits": "MHgwMGFiYTE4MA==",
        "boot-args": "",
        "oblit-begins": "T2JsaXRUeXBlOiBPYmxpdGVyYXRlRGF0YVBhcnRpdGlvbi4gUmVhc29uOiBpT1MgU2V0dXAgQXNzaXN0YW50IFN0YXJ0IE92ZXI=",
        "obliteration": "aGFuZGxlX21lc3NhZ2U6IE9ibGl0ZXJhdGlvbiBDb21wbGV0ZQo=",
        "usbcfwflasherResult": "Tm8gZXJyb3Jz"
    }
```

base64解出来的样子，看起来也没什么关系
```text
auto-boot: true
backlight-level: 1515
backlight-nits: 0x00aba180
boot-args: 
oblit-begins: OblitType: ObliterateDataPartition. Reason: iOS Setup Assistant Start Over
obliteration: handle_message: Obliteration Complete
usbcfwflasherResult: No errors
```

还有一种办法就是，直接根据ModelNumber 去查 `tidevice info -k ModelNumber`
然后去网站 <https://www.theiphonewiki.com/wiki/Models> 再查找对应的内存大小，就是可能有点麻烦

linmscat, 2021-04-02:
> 你说的是RAM size，目前看来好像也没有能用的接口
> `tidevice info --json`倒是有个看起来相关的信息
> 
> ```json
> {
>   "NonVolatileRAM": {
>         "auto-boot": "dHJ1ZQ==",
>         "backlight-level": "MTUxNQ==",
>         "backlight-nits": "MHgwMGFiYTE4MA==",
>         "boot-args": "",
>         "oblit-begins": "T2JsaXRUeXBlOiBPYmxpdGVyYXRlRGF0YVBhcnRpdGlvbi4gUmVhc29uOiBpT1MgU2V0dXAgQXNzaXN0YW50IFN0YXJ0IE92ZXI=",
>         "obliteration": "aGFuZGxlX21lc3NhZ2U6IE9ibGl0ZXJhdGlvbiBDb21wbGV0ZQo=",
>         "usbcfwflasherResult": "Tm8gZXJyb3Jz"
>     }
> ```
> 
> base64解出来的样子，看起来也没什么关系
> 
> ```
> auto-boot: true
> backlight-level: 1515
> backlight-nits: 0x00aba180
> boot-args: 
> oblit-begins: OblitType: ObliterateDataPartition. Reason: iOS Setup Assistant Start Over
> obliteration: handle_message: Obliteration Complete
> usbcfwflasherResult: No errors
> ```
> 
> 还有一种办法就是，直接根据ModelNumber 去查 `tidevice info -k ModelNumber`
> 然后去网站 https://www.theiphonewiki.com/wiki/Models 再查找对应的内存大小，就是可能有点麻烦

这上面好像也只有硬盘大小，查不到对应机型运存的大小

codeskyblue, 2021-04-08:
这个链接上有 https://socialcompare.com/en/comparison/apple-iphone-product-line-comparison

[image]