alibaba/tidevice #61: bug-执行instruments.app_process_list方法会崩溃

linmscat, 2021-04-02:
代码如下：
app_list = self._dev.instruments.app_list()
processes = next(self._dev.instruments.app_process_list(app_list))
print(processes)
执行报如下错误：
..\..\..\..\venv\lib\site-packages\tidevice\_instruments.py:709: in app_process_list
    info = exefile2appinfo(app_exepath, app_infos)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

exe_abspath = '/System/Library/PrivateFrameworks/MapsSupport.framework/geocorrectiond'
app_infos = [{'BundlePath': '/private/var/containers/Bundle/Application/045FE01F-8BE8-466E-9D5B-21AC78B8BE07/Weather.app/PlugIns/W...eIdentifier': 'com.apple.DiagnosticsService', 'ContainerBundlePath': '/Applications/DiagnosticsService.app', ...}, ...]

    def exefile2appinfo(exe_abspath: str, app_infos: List[dict]):
        for info in app_infos:
>           path = info['Path'] + "/" + info['CFBundleExecutable']
E           KeyError: 'Path'

..\..\..\..\venv\lib\site-packages\tidevice\_instruments.py:699: KeyError
@codeskyblue

codeskyblue, 2021-04-02:
你把 for info in app_infos:  中的 `info` 打出来看看是啥

linmscat, 2021-04-02:
> 你把 for info in app_infos: 中的 `info` 打出来看看是啥

打断点是info字典里面没有‘Path’这个key，导致info['Path']这里取值发生崩溃，你也可以本地运行下，必现的bug @codeskyblue

commit 905c3fb7 (2022-05-09): fix #61
closed 2022-05-09 by codeskyblue