alibaba/tidevice #64: ssl.SSLError: [SSL: CA_MD_TOO_WEAK]

thlengane, 2021-04-08:
Hi,

I have the devices connected on Ubuntu 20.04.2 LTS and running python 3.8.5, tidevice 0.3.5. When I run tidevice list, it shows the connected two devices but it throws ssl error on any other tidevice command, see below:-

UBUNTUX@HOSTPC:~$ tidevice list
List of apple devices attached
2ad8cf567fa8c3444e58010c874d3b817e67cc87 UBUNTUX’s iPhone
XXX08020-001648E40ED8003A iPhone

UBUNTUX@HOSTPC:~$ tidevice -u XXX08020-001648E40ED8003A applist
Traceback (most recent call last):
  File "/home/UBUNTUX/.local/bin/tidevice", line 8, in <module>
    sys.exit(main())
  File "/home/UBUNTUX/.local/lib/python3.8/site-packages/tidevice/__main__.py", line 627, in main
    actions[args.subparser](args)
  File "/home/UBUNTUX/.local/lib/python3.8/site-packages/tidevice/__main__.py", line 256, in cmd_applist
    for info in d.installation.iter_installed():
  File "/home/UBUNTUX/.local/lib/python3.8/site-packages/tidevice/_device.py", line 537, in installation
    conn = self.start_service(Installation.SERVICE_NAME)
  File "/home/UBUNTUX/.local/lib/python3.8/site-packages/tidevice/_device.py", line 454, in start_service
    return self._unsafe_start_service(name)
  File "/home/UBUNTUX/.local/lib/python3.8/site-packages/tidevice/_device.py", line 462, in _unsafe_start_service
    with self.create_session() as s:
  File "/usr/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/home/UBUNTUX/.local/lib/python3.8/site-packages/tidevice/_device.py", line 343, in create_session
    s.switch_to_ssl(self.ssl_pemfile_path)
  File "/home/UBUNTUX/.local/lib/python3.8/site-packages/tidevice/_safe_socket.py", line 73, in switch_to_ssl
    ssock = ssl.wrap_socket(self._sock,
  File "/usr/lib/python3.8/ssl.py", line 1402, in wrap_socket
    context.load_cert_chain(certfile, keyfile)
**ssl.SSLError: [SSL: CA_MD_TOO_WEAK] ca md too weak (_ssl.c:4022)**

Could there be something I am doing wrong?

thlengane, 2021-04-08:
I did this and it solved the error. Not sure if this is a bug on tidevice or just some certificate update needed.

**UBUNTUX@HOSTPC:~$** sudo vim /etc/ssl/openssl.cnf

Add this line at the top:
openssl_conf = openssl_init

And add these lines at the end:
[openssl_init]
ssl_conf = ssl_sect

[ssl_sect]
system_default = system_default_sect

[system_default_sect]
CipherString = DEFAULT@SECLEVEL=1

**UBUNTUX@HOSTPC:~$** export OPENSSL_CONF=/etc/ssl/openssl.cnf

**UBUNTUX@HOSTPC:~$** tidevice -u XXX08020-001648E40ED8003A applist
poc.org.internal.WebDriverAgentRunner.xctrunner WebDriverAgentRunner-Runner 1.0
poc.org.internal.IntegrationTests.xctrunner IntegrationTests_3-Runner 1.0

codeskyblue, 2021-04-08:
I searched in stackoverflow, found one relative issue: https://stackoverflow.com/questions/52218876/how-to-fix-ssl-issue-ssl-ctx-use-certificate-ca-md-too-weak-on-python-zeep

But it should not happend in tidevice. the logic is using sha1 for cert resign. em ....

thlengane, 2021-04-08:
> I searched in stackoverflow, found one relative issue: https://stackoverflow.com/questions/52218876/how-to-fix-ssl-issue-ssl-ctx-use-certificate-ca-md-too-weak-on-python-zeep
> 
> But it should not happend in tidevice. the pair code is using sha1 for cert resign (not md5). em ....

True, for me the trick seems to be this line **CipherString = DEFAULT@SECLEVEL=1** in the openssl.ncf file but when I read through, the latest linux os seems to be the issue.

HEZI0427, 2021-10-16:
@codeskyblue you can see this fixed issue，https://github.com/adrienverge/openfortivpn/issues/682
[image]
[image]

tidevice use ssl PROTOCOL_TLSv1_2，but openssl default ssl secur level up to 2 from 20.04

felixruan, 2021-11-04:
I know it's much better to use another certificate but sometimes there is no choise:
```
        # context = ssl.SSLContext(ssl.PROTOCOL_TLSv1_2)
        context = ssl.SSLContext()
        context.set_ciphers('ALL:@SECLEVEL=0')
```
I know it's very rude solution. Use it as start point

https://github.com/ma1co/Sony-PMCA-RE/issues/129#issuecomment-575912467
——粗鲁的的方法但可以解决Ubuntu20.04下OpenSSL问题

codeskyblue, 2022-05-09:
@felixruan thanks, fixed use this way.

closed 2022-05-09 by codeskyblue