alibaba/tidevice #65: 运行 tidevice applist报错

tinyleelee, 2021-04-09:
可以安装 运行报错
Traceback (most recent call last):
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\hhh\AppData\Local\Programs\Python\Python39\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\site-packages\tidevice\__main__.py", line 627, in main
    actions[args.subparser](args)
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\site-packages\tidevice\__main__.py", line 256, in cmd_applist
    for info in d.installation.iter_installed():
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 537, in installation
    conn = self.start_service(Installation.SERVICE_NAME)
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 454, in start_service
    return self._unsafe_start_service(name)
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 462, in _unsafe_start_service
    with self.create_session() as s:
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\contextlib.py", line 117, in __enter__
    return next(self.gen)
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 319, in create_session
    "HostID": self.pair_record['HostID'],
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 131, in pair_record
    self.handshake()
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 240, in handshake
    self._pair_record = self._read_pair_record()
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 167, in _read_pair_record
    data = self._usbmux.send_recv(payload)
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\site-packages\tidevice\_usbmux.py", line 49, in send_recv
    self._check(recv_data)
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\site-packages\tidevice\_usbmux.py", line 84, in _check
    raise MuxReplyError(data['Number'])
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\site-packages\tidevice\exceptions.py", line 14, in __init__
    self.reply_code = UsbmuxReplyCode(number)
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\enum.py", line 315, in __call__
    return cls.__new__(cls, value)
  File "c:\users\hhh\appdata\local\programs\python\python39\lib\enum.py", line 611, in __new__
    raise ve_exc
ValueError: 22 is not a valid UsbmuxReplyCode

electricbubble, 2021-04-09:
`tidevice` 的版本提供下

```bash
$ tidevice version
```

tinyleelee, 2021-04-09:
> `tidevice` 的版本提供下
> 
> ```shell
> $ tidevice version
> ```

0.3.5

a805791737, 2021-05-13:
运行 tidevice list报错
List of apple devices attached
Traceback (most recent call last):
  File "d:\python\lib\runpy.py", line 198, in _run_module_as_main
    "__main__", mod_spec)
  File "d:\python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "D:\python\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "d:\python\lib\site-packages\tidevice\__main__.py", line 687, in main
    actions[args.subparser](args)
  File "d:\python\lib\site-packages\tidevice\__main__.py", line 82, in cmd_list
    for dinfo in um.device_list():
  File "d:\python\lib\site-packages\tidevice\_usbmux.py", line 75, in device_list
    data = self.send_recv(payload)
  File "d:\python\lib\site-packages\tidevice\_usbmux.py", line 46, in send_recv
    with self.create_connection() as s:
  File "d:\python\lib\site-packages\tidevice\_usbmux.py", line 43, in create_connection
    return PlistSocket(self.__address, self._next_tag())
  File "d:\python\lib\site-packages\tidevice\_safe_socket.py", line 95, in __init__
    super().__init__(addr)
  File "d:\python\lib\site-packages\tidevice\_safe_socket.py", line 48, in __init__
    self._sock.connect(addr)
ConnectionRefusedError: [WinError 10061] 由于目标计算机积极拒绝，无法连接。

codeskyblue, 2021-05-18:
@a805791737 itunes没装

closed 2021-05-18 by codeskyblue