alibaba/tidevice #66: 求助-WDA经常出现崩溃

linmscat, 2021-04-12:
File "selenium\webdriver\remote\webdriver.py", line 157, in __init__
  File "appium\webdriver\webdriver.py", line 225, in start_session
  File "selenium\webdriver\remote\webdriver.py", line 321, in execute
  File "selenium\webdriver\remote\errorhandler.py", line 242, in check_response
selenium.common.exceptions.WebDriverException: Message: An unknown server-side error occurred while processing the command. @codeskyblue

linmscat, 2021-04-12:
App paths need to be absolute or an URL to a compressed app file: EPERM: operation not permitted, rename

codeskyblue, 2021-04-12:
你把tidevice的失败日志也贴出来

linmscat, 2021-04-12:
> 你把tidevice的失败日志也贴出来

Traceback (most recent call last):
  File "meter\thread\standard_thread.py", line 45, in run
  File "meter\thread\standard_thread.py", line 104, in _fire_thread_started
  File "mods\meter\listener\vuser_thread_listener.py", line 42, in started
  File "mods\meter\listener\vuser_thread_listener.py", line 77, in _build_driver
  File "driver\ios\ios_appium.py", line 35, in __init__
  File "device\ios\ios_device.py", line 59, in make_driver
  File "appium\webdriver\webdriver.py", line 151, in __init__
  File "selenium\webdriver\remote\webdriver.py", line 157, in __init__
  File "appium\webdriver\webdriver.py", line 225, in start_session
  File "selenium\webdriver\remote\webdriver.py", line 321, in execute
  File "selenium\webdriver\remote\errorhandler.py", line 242, in check_response
selenium.common.exceptions.WebDriverException: Message: An unknown server-side error occurred while processing the command. Original error: Bad app: D:\Users\yl1150\AppData\Roaming\apprunner\log\packages\Yealink-Meeting-40.0.32+qa.ipa. App paths need to be absolute or an URL to a compressed app file: EPERM: operation not permitted, rename 'D:\Users\yl1150\AppData\Local\Temp\2021312-15604-w1xyl8.2jk1\Payload\Odin-YMS.app' -> 'D:\Users\yl1150\AppData\Local\Temp\2021312-15604-a1bguq.x6dx\Odin-YMS.app'

linmscat, 2021-04-12:
tidevice的失败日志哪里可以导出

codeskyblue, 2021-04-12:
你没用命令行启动tidevice吗？

linmscat, 2021-04-12:
有的