alibaba/tidevice #68: tidevice applist app longversion

thlengane, 2021-04-14:
tidevice applist displays the list of apps and the ShortVersion ie. **6.0.4**, is there a way to display a long version ie. **6.0.4-35 (B5634)**? 

If not, can I please put this as a feature request?

electricbubble, 2021-04-14:
```python
from tidevice._usbmux import Usbmux
from tidevice import Device

if __name__ == "__main__":
    um = Usbmux()

    d = um.device_list()[0]
    d = Device(udid=d["SerialNumber"])
    bundleId = "com.tencent.xin"
    result = d.installation.lookup(bundleId)
    for k in result:
        if str(k).lower().find("version") >= 0:
            print("{:<30}\t{}".format(k, result[k]))

```

```bash
DTPlatformVersion             	14.4
CFBundleShortVersionString    	8.0.4
CFBundleInfoDictionaryVersion 	6.0
WeChatBundleVersion           	8.0.4.34
CFBundleVersion               	8.0.4.34
MinimumOSVersion              	13.0
CFBundleNumericVersion        	0
```

thlengane, 2021-04-14:
Perfect man, thanks. Working 100%. Appreciated.