alibaba/tidevice #69: wifi 连接的设备是不支持的对吧？

zangtian2, 2021-04-14:
通过wifi 连接的设备是不支持的对吧？

electricbubble, 2021-04-14:
苹果还没开放支持无线调试吧

zangtian2, 2021-04-14:
可以的，在Xcode的设备管理中开启。目前就这么跑的。

electricbubble, 2021-04-14:
求告知如何操作

[image]

zangtian2, 2021-04-14:
[image]

electricbubble, 2021-04-14:
代码里只筛选了 `USB`

https://github.com/alibaba/taobao-iphone-device/blob/16c185b90fd499747e9f97e1a4ea4977f0916d0f/tidevice/_usbmux.py#L77

zangtian2, 2021-04-14:
好的，谢谢，我本地改一下试试

codeskyblue, 2022-05-09:
目前指定udid也可以用。试试最新的版本

dddzname, 2022-07-06:
同问这个问题，指定udid 用wify链接的话要用什么命令

codeskyblue, 2022-07-19:
这个其实不太好用，偶尔用用还行

Haif-07, 2023-07-03:
> > 目前指定udid也可以用。尝试最新的版本
> 
> 指定udid用wifi连接的话用什么命令啊，查遍了文档和问题也没找到

请问解决了，目前遇到同样的问题。找不到什么指令打开wifi调试。在Windows上