alibaba/tidevice #70: 求助-facebook-WDA遇到加载列表性的控件就会卡主，很高概率出现WDA的socket掉线

linmscat, 2021-04-15:
问题：比如自动化操作iOS的textfield控件，往textfield中输入key，这时候会有相应的搜索结果展示在tableview控件中，此时会导致WDA代理响应的时间特别慢，像是整个工具卡主的感觉，最后极其容易出现WDA的socket断开的现象
wda断开堆栈：
File "F:\GitLab\apprunner\venv\lib\site-packages\selenium\webdriver\remote\webelement.py", line 80, in click
    self._execute(Command.CLICK_ELEMENT)
  File "F:\GitLab\apprunner\venv\lib\site-packages\selenium\webdriver\remote\webelement.py", line 633, in _execute
    return self._parent.execute(command, params)
  File "F:\GitLab\apprunner\venv\lib\site-packages\selenium\webdriver\remote\webdriver.py", line 321, in execute
    self.error_handler.check_response(response)
  File "F:\GitLab\apprunner\venv\lib\site-packages\appium\webdriver\errorhandler.py", line 31, in check_response
    raise wde
  File "F:\GitLab\apprunner\venv\lib\site-packages\appium\webdriver\errorhandler.py", line 26, in check_response
    super().check_response(response)
  File "F:\GitLab\apprunner\venv\lib\site-packages\selenium\webdriver\remote\errorhandler.py", line 242, in check_response
    raise exception_class(message, screen, stacktrace)
selenium.common.exceptions.WebDriverException: Message: An unknown server-side error occurred while processing the command. Original error: Could not proxy. Proxy error: Could not proxy command to the remote server. Original error: socket hang up
@codeskyblue @zy02636 @alibaba-oss @373743261 @pengchenglin

codeskyblue, 2021-04-16:
估计是遍历的元素太多，结果超时了。

linmscat, 2021-04-16:
> 估计是遍历的元素太多，结果超时了。

这个有办法优化吗，目前有遇到列表性控件的用例都过不下去

weikilla, 2021-04-20:
> > 估计是遍历的元素太多，结果超时了。
> 
> 这个有办法优化吗，目前有遇到列表性控件的用例都过不下去

把超时时间调大看看吧...https://blog.csdn.net/Pumpkinman/article/details/114737132

codeskyblue, 2021-04-20:
参考 @weikilla 减少层级感觉也可以。下面修改设置的方法参考一下
```python
c = wda.USBClient()
c.appium_settings({
  "snapshotMaxDepth": 30, # 默认是50
  ”snapshotTimeout": 60, # 默认15
})
```

linmscat, 2021-04-21:
> 参考 @weikilla 减少层级感觉也可以。下面修改设置的方法参考一下
> 
> ```python
> c = wda.USBClient()
> c.appium_settings({
>   "snapshotMaxDepth": 30, # 默认是50
>   ”snapshotTimeout": 60, # 默认15
> })
> ```
感谢两位大佬，我试试再回复使用结果

linmscat, 2021-04-21:
> 参考 @weikilla 减少层级感觉也可以。下面修改设置的方法参考一下
> 
> ```python
> c = wda.USBClient()
> c.appium_settings({
>   "snapshotMaxDepth": 30, # 默认是50
>   ”snapshotTimeout": 60, # 默认15
> })
> ```

结果会出现：
[D 210421 10:53:08 _relay:36] handle stream from: ('127.0.0.1', 49501)
[E 210421 10:53:08 _relay:46] connect to device error: UsbmuxReplyCode.ConnectionRefused
导致xctrunner quited @codeskyblue

linmscat, 2021-04-22:
> 结果会出现：
> [D 210421 10:53:08 _relay:36] handle stream from: ('127.0.0.1', 49501)
> [E 210421 10:53:08 _relay:46] connect to device error: UsbmuxReplyCode.ConnectionRefused
> 导致xctrunner quited @codeskyblue

这个问题有办法解决吗，难道ios自动化测不了列表相关的UI界面？ @codeskyblue

electricbubble, 2021-04-22:
建议使用 `Xcode` 试试这个场景

linmscat, 2021-04-22:
> 建议使用 `Xcode` 试试这个场景
> 
> > 🤫 或者试试我写的 cli 版本

1、使用Xcode test对应的WDA？这不就回退到之前的那种方式
2、你写的cli 版本是什么？怎么使用呢

electricbubble, 2021-04-22:
> 1. 使用Xcode test对应的WDA？这不就回退到之前的那种方式

就是用以前只能用 `macOS` 的方式, 看看这个问题是否也会在 `Xcode` 上出现

linmscat, 2021-04-22:
我怀疑是tidevice的bug，以前用过xcode，只会卡主，但不会导致xctrunner quited

electricbubble, 2021-04-22:
能提供下 `app` 和 可复现问题的 `脚本` 吗...

TangSir61, 2021-12-09:
京东，购物车页面，加200个种类商品，就可以复现