alibaba/tidevice #71: Crash-tidevice崩溃

linmscat, 2021-04-16:
崩溃堆栈：
File "F:\GitLab\apprunner\venv\lib\site-packages\tidevice\_device.py", line 820, in instruments
    return self.connect_instruments()
  File "F:\GitLab\apprunner\venv\lib\site-packages\tidevice\_device.py", line 809, in connect_instruments
    conn = self.start_service(
  File "F:\GitLab\apprunner\venv\lib\site-packages\tidevice\_device.py", line 454, in start_service
    return self._unsafe_start_service(name)
  File "F:\GitLab\apprunner\venv\lib\site-packages\tidevice\_device.py", line 462, in _unsafe_start_service
    with self.create_session() as s:
  File "D:\Users\yl1150\AppData\Local\Programs\Python\Python38\lib\contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "F:\GitLab\apprunner\venv\lib\site-packages\tidevice\_device.py", line 343, in create_session
    s.switch_to_ssl(self.ssl_pemfile_path)
  File "F:\GitLab\apprunner\venv\lib\site-packages\tidevice\_safe_socket.py", line 73, in switch_to_ssl
    ssock = ssl.wrap_socket(self._sock,
  File "D:\Users\yl1150\AppData\Local\Programs\Python\Python38\lib\ssl.py", line 1405, in wrap_socket
    return context.wrap_socket(
  File "D:\Users\yl1150\AppData\Local\Programs\Python\Python38\lib\ssl.py", line 500, in wrap_socket
    return self.sslsocket_class._create(
  File "D:\Users\yl1150\AppData\Local\Programs\Python\Python38\lib\ssl.py", line 1040, in _create
    self.do_handshake()
  File "D:\Users\yl1150\AppData\Local\Programs\Python\Python38\lib\ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error
@codeskyblue

codeskyblue, 2021-04-16:
是不是手机弹出信任窗口，你没点信任？

linmscat, 2021-04-16:
> 是不是手机弹出信任窗口，你没点信任？

这个是挂机5,6个小时才出现的崩溃，从录屏上看，系统没有弹出什么权限弹窗