alibaba/tidevice #72: wda启动失败已经挂载了镜像 DeveloperImage already mounted

Tester-957, 2021-04-16:
Could not start service: com.apple.mobile.house_arrest! Remember that you have to mount the Developer disk image on your device

`[D 210416 11:30:55 _wdaproxy:116] [00008101-001240463E91003A] launch WDA
[I 210416 11:30:56 _device:990] BundleID: com.facebook.WebDriverAgentRunner12aw1222.xctrunner
[I 210416 11:30:56 _device:992] DeviceIdentifier: 00008101-001240463E91003A
[I 210416 11:30:56 _device:834] SignIdentity: 'Apple Development: lizhi17767@163.com (3L73AL2395)'
[I 210416 11:30:56 _device:840] CFBundleExecutable: WebDriverAgentRunner-Runner
[I 210416 11:30:56 _device:633] DeveloperImage already mounted
Traceback (most recent call last):
  File "c:\users\egonyuan\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 454, in start_service
    return self._unsafe_start_service(name)
  File "c:\users\egonyuan\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 469, in _unsafe_start_service
    raise MuxServiceError("Could not start service: {0}! {2}".format(
tidevice.exceptions.MuxServiceError: Could not start service: com.apple.mobile.house_arrest! Remember that you have to mount the Developer disk image on your device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\egonyuan\appdata\local\programs\python\python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\egonyuan\appdata\local\programs\python\python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\users\egonyuan\appdata\local\programs\python\python39\lib\site-packages\tidevice\__main__.py", line 632, in <module>
    main()
  File "c:\users\egonyuan\appdata\local\programs\python\python39\lib\site-packages\tidevice\__main__.py", line 627, in main
    actions[args.subparser](args)
  File "c:\users\egonyuan\appdata\local\programs\python\python39\lib\site-packages\tidevice\__main__.py", line 227, in cmd_xctest
    d.xctest(args.bundle_id,
  File "c:\users\egonyuan\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 1063, in xctest
    pid = self._launch_app_runner(bundle_id, session_identifier,
  File "c:\users\egonyuan\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 851, in _launch_app_runner
    fsync = self.app_sync(bundle_id)
  File "c:\users\egonyuan\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 528, in app_sync
    conn = self.start_service(LockdownService.MobileHouseArrest)
  File "c:\users\egonyuan\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 459, in start_service
    return self._unsafe_start_service(name)
  File "c:\users\egonyuan\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 469, in _unsafe_start_service
    raise MuxServiceError("Could not start service: {0}! {2}".format(
tidevice.exceptions.MuxServiceError: Could not start service: com.apple.mobile.house_arrest! Remember that you have to mount the Developer disk image on your device
[E 210416 11:30:57 _wdaproxy:124] [00008101-001240463E91003A] wda started failed`

codeskyblue, 2021-04-16:
house_arrest 是苹果的文件服务，暂时我也不知道怎么整。先重启一下手机，一般都能解决。

codeskyblue, 2023-10-19:
暂时关了

closed 2023-10-19 by codeskyblue