alibaba/tidevice #75: TLS/SSL connection has been closed (EOF)

li8911255, 2021-04-19:
WDA 启动失败，完整日志如下：
[I 210419 11:55:02 _device:990] BundleID: com.facebook.WebDriverAgentRunner.xctrunner
[I 210419 11:55:02 _device:992] DeviceIdentifier: 00008020-00146D593669002E
[I 210419 11:55:02 _device:834] SignIdentity: 'Apple Development: xuefei yu (3832536FLT)'
[I 210419 11:55:02 _device:840] CFBundleExecutable: WebDriverAgentRunner-Runner
Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.9/3.9.2_4/Frameworks/Python.framework/Versions/3.9/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/Cellar/python@3.9/3.9.2_4/Frameworks/Python.framework/Versions/3.9/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Users/leo/Code/taobao-iphone-device/tidevice/__main__.py", line 632, in <module>
    main()
  File "/Users/leo/Code/taobao-iphone-device/tidevice/__main__.py", line 627, in main
    actions[args.subparser](args)
  File "/Users/leo/Code/taobao-iphone-device/tidevice/__main__.py", line 227, in cmd_xctest
    d.xctest(args.bundle_id,
  File "/Users/leo/Code/taobao-iphone-device/tidevice/_device.py", line 1063, in xctest
    pid = self._launch_app_runner(bundle_id, session_identifier,
  File "/Users/leo/Code/taobao-iphone-device/tidevice/_device.py", line 860, in _launch_app_runner
    channel = conn.make_channel(InstrumentsService.ProcessControl)
  File "/Users/leo/Code/taobao-iphone-device/tidevice/_instruments.py", line 313, in make_channel
    result = self.call_message(0, '_requestChannelWithCode:identifier:', aux)
  File "/Users/leo/Code/taobao-iphone-device/tidevice/_instruments.py", line 354, in call_message
    _id = self.send_dtx_message(channel,
  File "/Users/leo/Code/taobao-iphone-device/tidevice/_instruments.py", line 390, in send_dtx_message
    self.sendall(data)
  File "/Users/leo/Code/taobao-iphone-device/tidevice/_safe_socket.py", line 66, in sendall
    return self._sock.sendall(data)
  File "/usr/local/Cellar/python@3.9/3.9.2_4/Frameworks/Python.framework/Versions/3.9/lib/python3.9/ssl.py", line 1204, in sendall
    v = self.send(byte_view[count:])
  File "/usr/local/Cellar/python@3.9/3.9.2_4/Frameworks/Python.framework/Versions/3.9/lib/python3.9/ssl.py", line 1173, in send
    return self._sslobj.write(data)
ssl.SSLZeroReturnError: TLS/SSL connection has been closed (EOF) (_ssl.c:2472)

Nnnnnnnnnical, 2022-01-05:
同遇到，啥都没做，之前还是可以的，跑着跑着突然出现

Nnnnnnnnnical, 2022-01-05:
重启手机解决.....

codeskyblue, 2022-03-24:
possible duplicated with https://github.com/alibaba/taobao-iphone-device/issues/148

closed 2022-07-26 by codeskyblue