alibaba/tidevice #78: 性能数据收集

pqs-fly, 2021-04-22:
tidevice perf -o fps,mem,cpu -B com.example.demo  ，这个命令现在还没有完成吗？

codeskyblue, 2021-05-12:
fixed in 0.4.6

lycfr, 2021-05-17:
``C:\Users\xxx>tidevice perf -B com.lyyd.demo
Stopped
Traceback (most recent call last):
  File "d:\program files\python\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "d:\program files\python\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "D:\Program Files\Python\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "d:\program files\python\lib\site-packages\tidevice\__main__.py", line 687, in main
    actions[args.subparser](args)
  File "d:\program files\python\lib\site-packages\tidevice\__main__.py", line 461, in cmd_perf
    signal.pause()
AttributeError: module 'signal' has no attribute 'pause'

C:\Users\xxx>tidevice -v
0.4.6

codeskyblue, 2021-05-17:
0.4.8再试试

lycfr, 2021-05-18:
@codeskyblue 
执行一会发现报错，程序堵住了
```
fps {'fps': 24, 'value': 24, 'timestamp': 1621306532555}
fps {'fps': 21, 'value': 21, 'timestamp': 1621306533559}
Exception in thread Thread-2:
Traceback (most recent call last):
  File "d:\program files\python\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "d:\program files\python\lib\threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "d:\program files\python\lib\site-packages\tidevice\_instruments.py", line 560, in _drain
    self._drain_single_message()
  File "d:\program files\python\lib\site-packages\tidevice\_instruments.py", line 611, in _drain_single_message
    self._reply_null(dtxm)
  File "d:\program files\python\lib\site-packages\tidevice\_instruments.py", line 518, in _reply_null
    message_id=m.message_id)
  File "d:\program files\python\lib\site-packages\tidevice\_instruments.py", line 392, in send_dtx_message
    self.sendall(data)
  File "d:\program files\python\lib\site-packages\tidevice\_safe_socket.py", line 66, in sendall
    return self._sock.sendall(data)
  File "d:\program files\python\lib\ssl.py", line 972, in sendall
    v = self.send(byte_view[count:])
  File "d:\program files\python\lib\ssl.py", line 941, in send
    return self._sslobj.write(data)
  File "d:\program files\python\lib\ssl.py", line 642, in write
    return self._sslobj.write(data)
OSError: [WinError 10014] 系统检测到在一个调用中尝试使用指针参数时的无效指针地址。
```

codeskyblue, 2021-05-18:
网上其他人提出的解决办法 https://www.cnblogs.com/vitas/p/9025857.html

解决办法： 用管理身份运行cmd，执行 `netsh winsock reset`  重启问题解决

lycfr, 2021-05-20:
问题还是会出现的，没法弄。就解析fps的数据出不来。其他如果程序没堵住的话，还是会打印出来的。如果堵住就没法输出了。有单独指定采集性能数据的指令吗？比如，我想单独采集内存或者CPU，或者同时采集某几个性能数据

linmscat, 2021-06-16:
> 问题还是会出现的，没法弄。就解析fps的数据出不来。其他如果程序没堵住的话，还是会打印出来的。如果堵住就没法输出了。有单独指定采集性能数据的指令吗？比如，我想单独采集内存或者CPU，或者同时采集某几个性能数据

這個問題怎么解决

FuChen1649, 2021-07-09:
能不能监控到app级别的耗电情况？

commit 25d2eb12 (2021-05-12): fix #78