alibaba/tidevice #79: wda+UNtest+htmltestrunner运行的自动化脚本的时候老是出现'device port:8100 is not ready'

name-jiang, 2021-04-23:
ft10_1: Traceback (most recent call last):
  File "D:\python3\lib\site-packages\urllib3\connectionpool.py", line 677, in urlopen
    chunked=chunked,
  File "D:\python3\lib\site-packages\urllib3\connectionpool.py", line 392, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "D:\python3\lib\http\client.py", line 1277, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "D:\python3\lib\http\client.py", line 1323, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "D:\python3\lib\http\client.py", line 1272, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "D:\python3\lib\http\client.py", line 1032, in _send_output
    self.send(msg)
  File "D:\python3\lib\http\client.py", line 972, in send
    self.connect()
  File "D:\python3\lib\site-packages\wda\requests_usbmux.py", line 54, in connect
    conn = _device.create_inner_connection(int(port))
  File "D:\python3\lib\site-packages\wda\usbmux.py", line 239, in create_inner_connection
    "device port:{} is not ready".format(_original_port))
wda.exceptions.MuxConnectError: device port:8100 is not ready

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\python3\lib\site-packages\requests\adapters.py", line 449, in send
    timeout=timeout
  File "D:\python3\lib\site-packages\urllib3\connectionpool.py", line 727, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "D:\python3\lib\site-packages\urllib3\util\retry.py", line 410, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "D:\python3\lib\site-packages\urllib3\packages\six.py", line 734, in reraise
    raise value.with_traceback(tb)
  File "D:\python3\lib\site-packages\urllib3\connectionpool.py", line 677, in urlopen
    chunked=chunked,
  File "D:\python3\lib\site-packages\urllib3\connectionpool.py", line 392, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "D:\python3\lib\http\client.py", line 1277, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "D:\python3\lib\http\client.py", line 1323, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "D:\python3\lib\http\client.py", line 1272, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "D:\python3\lib\http\client.py", line 1032, in _send_output
    self.send(msg)
  File "D:\python3\lib\http\client.py", line 972, in send
    self.connect()
  File "D:\python3\lib\site-packages\wda\requests_usbmux.py", line 54, in connect
    conn = _device.create_inner_connection(int(port))
  File "D:\python3\lib\site-packages\wda\usbmux.py", line 239, in create_inner_connection
    "device port:{} is not ready".format(_original_port))
urllib3.exceptions.ProtocolError: ('Connection aborted.', MuxConnectError('device port:8100 is not ready'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\app_test_ios\scripts\sgapp_home_userdefine_click.py", line 25, in setUp
    cold_startapp(client)
  File "D:\app_test_ios\base\sg_action.py", line 47, in cold_startapp
    client(name="跳过").click_exists(timeout=0.5)
  File "D:\python3\lib\site-packages\wda\__init__.py", line 1529, in click_exists
    e = self.get(timeout=timeout, raise_error=False)
  File "D:\python3\lib\site-packages\wda\__init__.py", line 1469, in get
    elems = self.find_elements()
  File "D:\python3\lib\site-packages\wda\__init__.py", line 1444, in find_elements
    for element_id in self.find_element_ids():
  File "<decorator-gen-8>", line 2, in find_element_ids
  File "D:\python3\lib\site-packages\retry\api.py", line 74, in retry_decorator
    logger)
  File "D:\python3\lib\site-packages\retry\api.py", line 33, in __retry_internal
    return f()
  File "D:\python3\lib\site-packages\wda\__init__.py", line 1436, in find_element_ids
    return self._wdasearch('class chain', chain)
  File "D:\python3\lib\site-packages\wda\__init__.py", line 1384, in _wdasearch
    'value': value
  File "D:\python3\lib\site-packages\wda\utils.py", line 47, in _inner
    return fn(*args, **kwargs)
  File "D:\python3\lib\site-packages\wda\__init__.py", line 454, in _fetch
    response = httpdo(url, method, data, timeout)
  File "D:\python3\lib\site-packages\wda\__init__.py", line 124, in httpdo
    return _unsafe_httpdo(url, method, data, timeout)
  File "D:\python3\lib\site-packages\wda\__init__.py", line 154, in _unsafe_httpdo
    timeout=timeout)
  File "D:\python3\lib\site-packages\requests\sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "D:\python3\lib\site-packages\requests\sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "D:\python3\lib\site-packages\requests\adapters.py", line 498, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', MuxConnectError('device port:8100 is not ready'))

codeskyblue, 2021-04-28:
你的先把xctest启动起来再说啊

name-jiang, 2021-04-28:
运行过程中断，不知道是什么原因，发现大多时候都是在点击摸个元素，然后tidevice就突然重新启动了

codeskyblue, 2021-04-28:
@name-jiang 你是在测试什么应用，我昨天试了下抖音这种应用点击也会卡死掉。不是tidevice故意的，是WDA不响应了。

name-jiang, 2021-04-28:
> @ name-jiang你是在测试什么应用，我昨天试了下抖音这种应用点击也会卡死掉。不是tidevice故意的，是WDA不响应了。

我随便找的应用试验的，发现执行10条case的时候，tidevice就会重启，造成wda无法连接，需要重新启动wda。也不知道什么原因造成的

yangeren, 2021-06-11:
> tidevice

我也发现不是很稳定，出现一次与wda的连接超时，就会有一段时间无法使用，tidevice就偿试重连，过段时间就又可以了。

帖一下tidevice的日志：
[I 210611 14:21:58 _device:1003] Start execute test plan with IDE version: 29
[I 210611 14:21:58 _device:913] WebDriverAgent start successfully
[D 210611 14:22:31 _wdaproxy:93] [f26812b84ac67cc567707cb6a679f8ba37095936] WDA is not response, check again after 10s
[D 210611 14:24:16 _wdaproxy:93] [f26812b84ac67cc567707cb6a679f8ba37095936] WDA is not response, check again after 10s
[I 210611 14:24:29 _wdaproxy:97] [f26812b84ac67cc567707cb6a679f8ba37095936] WDA confirmed not running
[I 210611 14:24:29 _wdaproxy:106] [f26812b84ac67cc567707cb6a679f8ba37095936] WDA keeper stopped
[I 210611 14:24:29 _wdaproxy:145] [f26812b84ac67cc567707cb6a679f8ba37095936] WDA stopped for the 19st time, running 2.5 minutes
[D 210611 14:24:29 _wdaproxy:118] [f26812b84ac67cc567707cb6a679f8ba37095936] launch WDA
[I 210611 14:24:30 _device:956] BundleID: com.facebook.WebDriverAgentRunner.xctrunner
[I 210611 14:24:30 _device:959] ProductVersion: 12.1