alibaba/tidevice #81: tidevice 0.4.0版本 使用tidevice -u（设备 udid）wdaproxy -B（wda 的 bundleId） --port 8100报错

duanhonggo, 2021-04-23:
错误是python引用问题，好像是循环引用导致的，后续我再贴报错信息

duanhonggo, 2021-04-23:
[D 210423 15:19:45 _wdaproxy:116] [00008020-000679D90EC3002E] launch WDA
Exception in thread Thread-1:
Traceback (most recent call last):
  File "d:\program files\python38\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "d:\program files\python38\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "d:\program files\python38\lib\site-packages\tidevice\_wdaproxy.py", line 207, in _wrapped_func
    return self._func(*args)
  File "d:\program files\python38\lib\site-packages\tidevice\_wdaproxy.py", line 123, in _keep_wda_running
    if not self._wait_ready(proc, stop_event):
  File "d:\program files\python38\lib\site-packages\tidevice\_wdaproxy.py", line 66, in _wait_ready
    if self._is_alive():
  File "d:\program files\python38\lib\site-packages\tidevice\_wdaproxy.py", line 41, in _is_alive
    from ._usbmux import MuxReplyError, UsbmuxReplyCode
ImportError: cannot import name 'MuxReplyError' from 'tidevice._usbmux' (d:\program files\python38\lib\site-packages\tidevice\_usbmux.py)

codeskyblue, 2021-04-28:
fixed