alibaba/tidevice #82: bug-点击控件引起WDA超时，概率较高

linmscat, 2021-04-23:
File "F:\GitLab\apprunner\venv\lib\site-packages\selenium\webdriver\remote\webelement.py", line 80, in click
    self._execute(Command.CLICK_ELEMENT)
  File "F:\GitLab\apprunner\venv\lib\site-packages\selenium\webdriver\remote\webelement.py", line 633, in _execute
    return self._parent.execute(command, params)
  File "F:\GitLab\apprunner\venv\lib\site-packages\selenium\webdriver\remote\webdriver.py", line 321, in execute
    self.error_handler.check_response(response)
  File "F:\GitLab\apprunner\venv\lib\site-packages\appium\webdriver\errorhandler.py", line 31, in check_response
    raise wde
  File "F:\GitLab\apprunner\venv\lib\site-packages\appium\webdriver\errorhandler.py", line 26, in check_response
    super().check_response(response)
  File "F:\GitLab\apprunner\venv\lib\site-packages\selenium\webdriver\remote\errorhandler.py", line 242, in check_response
    raise exception_class(message, screen, stacktrace)
selenium.common.exceptions.WebDriverException: Message: An unknown server-side error occurred while processing the command. Original error: Could not proxy. Proxy error: Could not proxy command to the remote server. Original error: timeout of 240000ms exceeded

huang1988519, 2021-04-25:
尝试修改下 wda 配置：static NSString* const SNAPSHOT_TIMEOUT = @"snapshotTimeout";
appium 也支持修改此项设置

codeskyblue, 2021-04-28:
有时候修改 snapshotMaxDepth 改成10以下，也可以，不过这个会导致dump元素缺失