alibaba/tidevice #85: 概率性WDA is not response

AllenLuosong, 2021-04-28:
请教下 如下报错是否和数据线有关，偶尔会出现wda无响应的异常
[D 210428 09:20:24 _relay:36] handle stream from: ('::1', 59532, 0, 0)
Exception in thread Thread-2:
[I 210428 09:22:29 _device:1114]Traceback (most recent call last):
  File "d:\program\python\lib\threading.py", line 932, in _bootstrap_inner
 xctrunner quited
[D 210428 09:22:29 _wdaproxy:91] [00008101-000D09C03A44001E] WDA is not response, check again after 10s

codeskyblue, 2021-04-28:
tidevice xctest --debug 查看一下原因

AllenLuosong, 2021-04-28:
是这个异常吗

[D 210428 16:13:48 _device:942] logProcess: 2021-04-28 16:13:49.444858+0800 WebDriverAgentRunner-Runner[514:69128] [User Defaults] Couldn't write values for keys (
        KeyboardAutocorrection
    ) in CFPrefsPlistSource<0x281ce3880> (Domain: com.apple.Preferences, User: kCFPreferencesCurrentUser, ByHost: No, Container: (null), Contents Need Refresh: No): setting preferences outside an application's container requires user-preference-write or file-write-data sandbox access

codeskyblue, 2021-05-07:
看起来是的，wda你看看给他发了那个命令导致死掉的

ashfish, 2021-05-11:
> 看起来是的，wda你看看给他发了哪个命令导致死掉的

页面上的元素过多的话就会导致WDA is not response, 哪里可以把超时时间从10秒调久一点?win端