alibaba/tidevice #88: 关于打包WDA.ipa问题（About Package WebDriverAgent.ipa）

ZhouYixun, 2021-05-08:
有比较详细的从github拉下来的webdriveragent开始的打包步骤吗，我是从npm i appium下来的，在appium-webdriveragent里面执行了Script等等步骤，现在也是用xcode build的方式顺利跑了好几个月。
现在想采用这个tidevice的方案，但是打包时出现很多问题，跟着这个指令
`xcodebuild build-for-testing -workspace WebDriverAgent.xcworkspace/ -scheme WebDriverAgent -sdk iphoneos -configuration Release -derivedDataPath /tmp/derivedDataPath`
会出现以下问题
`xcodebuild: error: 'WebDriverAgent.xcworkspace/' does not exist.`

之后我尝试添加目录为xcode的workspace
[image]
[image]

再执行，出现以下问题
`xcodebuild: error: The workspace named "WebDriverAgent" does not contain a scheme named "WebDriverAgent". The "-list" option can be used to find the names of the schemes in the workspace.`

最后将scheme改为WebDriverAgent-Runner，发现ipa执行不了

请问打包前的签名要用webdriveragent默认的，然后打完包再重签名？
还是打包前后都要重签名？因为跟之前传统方式有很多不一样的，希望作者能回答一下，感激不尽

ZhouYixun, 2021-05-08:
https://github.com/alibaba/taobao-iphone-device/blob/main/DEVELOP.md
我参考的是作者这个步骤来的，就是不行

electricbubble, 2021-05-08:
https://github.com/alibaba/taobao-iphone-device/issues/32

ZhouYixun, 2021-05-08:
I follow the https://github.com/alibaba/taobao-iphone-device/blob/main/DEVELOP.md#how-to-package-wdaipa
but I get
`Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/bin/tidevice", line 11, in <module>
    sys.exit(main())
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tidevice/__main__.py", line 669, in main
    actions[args.subparser](args)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tidevice/__main__.py", line 175, in cmd_parse
    ir.dump_info()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tidevice/_ipautil.py", line 70, in dump_info
    data = self.get_infoplist()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tidevice/_ipautil.py", line 55, in get_infoplist
    finfo = self.get_infoplist_zipinfo()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tidevice/_ipautil.py", line 26, in get_infoplist_zipinfo
    raise IPAError("Info.plist not found")
tidevice.exceptions.IPAError: Info.plist not found`

*.app contain IntegrationApp.app
my Release-iphoneos folder (after move file)
[image]
my Payload folder
[image]
xcode version: 14.2
webdriveragent version : newest

ZhouYixun, 2021-05-08:
It work now,thanks!

closed 2021-05-08 by ZhouYixun

codeskyblue, 2021-05-11:
congratulations