alibaba/tidevice #89: Mac重启后无法启动tidevice.exceptions.MuxReplyError: UsbmuxReplyCode.ConnectionRefused

whatRUsay, 2021-05-08:
之前一切正常可以运行，设备无故重启之后就无法启动，更换usb插口也无效，详细报错：
  Traceback (most recent call last):
  File "/usr/local/bin/tidevice", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/site-packages/tidevice/__main__.py", line 669, in main
    actions[args.subparser](args)
  File "/usr/local/lib/python3.9/site-packages/tidevice/__main__.py", line 227, in cmd_xctest
    d.xctest(args.bundle_id,
  File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 1002, in xctest
    x1 = self._connect_testmanagerd_lockdown()
  File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 801, in _connect_testmanagerd_lockdown
    conn = self.start_service(LockdownService.TestmanagerdLockdown)
  File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 454, in start_service
    return self._unsafe_start_service(name)
  File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 483, in _unsafe_start_service
    conn = self.create_inner_connection(data['Port'], _ssl=_ssl)
  File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 294, in create_inner_connection
    self._usbmux._check(data)
  File "/usr/local/lib/python3.9/site-packages/tidevice/_usbmux.py", line 84, in _check
    raise MuxReplyError(data['Number'])
tidevice.exceptions.MuxReplyError: UsbmuxReplyCode.ConnectionRefused

zhixideyu, 2021-09-28:
> 之前一切正常可以运行，设备无故重启之后就无法启动，更换usb插口也无效，详细报错： Traceback (most recent call last): File "/usr/local/bin/tidevice", line 8, in sys.exit(main()) File "/usr/local/lib/python3.9/site-packages/tidevice/**main**.py", line 669, in main actions[args.subparser](args) File "/usr/local/lib/python3.9/site-packages/tidevice/**main**.py", line 227, in cmd_xctest d.xctest(args.bundle_id, File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 1002, in xctest x1 = self._connect_testmanagerd_lockdown() File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 801, in _connect_testmanagerd_lockdown conn = self.start_service(LockdownService.TestmanagerdLockdown) File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 454, in start_service return self._unsafe_start_service(name) File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 483, in _unsafe_start_service conn = self.create_inner_connection(data['Port'], _ssl=_ssl) File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 294, in create_inner_connection self._usbmux._check(data) File "/usr/local/lib/python3.9/site-packages/tidevice/_usbmux.py", line 84, in _check raise MuxReplyError(data['Number']) tidevice.exceptions.MuxReplyError: UsbmuxReplyCode.ConnectionRefused

解决了 一样的问题

zhixideyu, 2021-09-28:
> > 之前一切正常可以运行，设备无故重启之后就无法启动，更换usb插口也无效，详细报错： Traceback (most recent call last): File "/usr/local/bin/tidevice", line 8, in sys.exit(main()) File "/usr/local/lib/python3.9/site-packages/tidevice/**main**.py", line 669, in main actions[args.subparser](args) File "/usr/local/lib/python3.9/site-packages/tidevice/**main**.py", line 227, in cmd_xctest d.xctest(args.bundle_id, File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 1002, in xctest x1 = self._connect_testmanagerd_lockdown() File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 801, in _connect_testmanagerd_lockdown conn = self.start_service(LockdownService.TestmanagerdLockdown) File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 454, in start_service return self._unsafe_start_service(name) File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 483, in _unsafe_start_service conn = self.create_inner_connection(data['Port'], _ssl=_ssl) File "/usr/local/lib/python3.9/site-packages/tidevice/_device.py", line 294, in create_inner_connection self._usbmux._check(data) File "/usr/local/lib/python3.9/site-packages/tidevice/_usbmux.py", line 84, in _check raise MuxReplyError(data['Number']) tidevice.exceptions.MuxReplyError: UsbmuxReplyCode.ConnectionRefused
> 
> 解决了吗？ 一样的问题

codeskyblue, 2021-10-12:
主要还是usbmuxd这个服务出了点问题

dev-xdyang, 2022-10-25:
开始安装的时候试过，可以正常启动app。但一段时间后再用就报这个错误，不知道啥问题