alibaba/tidevice #92: tidevice安装成功后,CMD输入tidevice list 报错怎么回事呀??

a805791737, 2021-05-13:
List of apple devices attached
Traceback (most recent call last):
  File "d:\python\lib\runpy.py", line 198, in _run_module_as_main
    "__main__", mod_spec)
  File "d:\python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "D:\python\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "d:\python\lib\site-packages\tidevice\__main__.py", line 687, in main
    actions[args.subparser](args)
  File "d:\python\lib\site-packages\tidevice\__main__.py", line 82, in cmd_list
    for dinfo in um.device_list():
  File "d:\python\lib\site-packages\tidevice\_usbmux.py", line 75, in device_list
    data = self.send_recv(payload)
  File "d:\python\lib\site-packages\tidevice\_usbmux.py", line 46, in send_recv
    with self.create_connection() as s:
  File "d:\python\lib\site-packages\tidevice\_usbmux.py", line 43, in create_connection
    return PlistSocket(self.__address, self._next_tag())
  File "d:\python\lib\site-packages\tidevice\_safe_socket.py", line 95, in __init__
    super().__init__(addr)
  File "d:\python\lib\site-packages\tidevice\_safe_socket.py", line 48, in __init__
    self._sock.connect(addr)
ConnectionRefusedError: [WinError 10061] 由于目标计算机积极拒绝，无法连接。

jongyu, 2021-05-13:
#50

codeskyblue, 2021-05-18:
装itunes就可以了。

closed 2021-05-18 by codeskyblue