alibaba/tidevice #93: 运行tidevice list报错

a805791737, 2021-05-13:
Microsoft Windows [版本 10.0.19041.928]
(c) Microsoft Corporation。保留所有权利。

C:\Users\D1M>tidevice list
List of apple devices attached
Traceback (most recent call last):
  File "d:\python\lib\runpy.py", line 198, in _run_module_as_main
    "__main__", mod_spec)
  File "d:\python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "D:\python\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "d:\python\lib\site-packages\tidevice\__main__.py", line 687, in main
    actions[args.subparser](args)
  File "d:\python\lib\site-packages\tidevice\__main__.py", line 82, in cmd_list
    for dinfo in um.device_list():
  File "d:\python\lib\site-packages\tidevice\_usbmux.py", line 75, in device_list
    data = self.send_recv(payload)
  File "d:\python\lib\site-packages\tidevice\_usbmux.py", line 46, in send_recv
    with self.create_connection() as s:
  File "d:\python\lib\site-packages\tidevice\_usbmux.py", line 43, in create_connection
    return PlistSocket(self.__address, self._next_tag())
  File "d:\python\lib\site-packages\tidevice\_safe_socket.py", line 95, in __init__
    super().__init__(addr)
  File "d:\python\lib\site-packages\tidevice\_safe_socket.py", line 48, in __init__
    self._sock.connect(addr)
ConnectionRefusedError: [WinError 10061] 由于目标计算机积极拒绝，无法连接。

jongyu, 2021-05-13:
#50 
搞不懂你一个问题为什么提交两遍,iTunes去苹果官网下载就可以了

a805791737, 2021-05-13:
不好意思,我以为第一次没提交上去.还以为装完tidevice后不用别的操作就可以直接连IOS设备了

 
#50

a805791737, 2021-05-13:
另外还是没明白这个报错和下载iTunes有什么直接的关系

 
#50

jongyu, 2021-05-13:
你安装就是了,最简单的理解,iPhone需要信任电脑,iTunes是苹果官方开的再Windows电脑上管理iPhone的软件, 深一点的就是需要借助里面东西和连接的iPhone进行通信

a805791737, 2021-05-13:
使用tidevice的安装命令，将 开发者证书重签名 的 WebDriverAgent.ipa 安装到iOS设备上。

想请问下这一步怎么实现?报错

C:\Users\D1M>tidevice -u 00008030-0008795C1183802E install https://example.org/example.ipa
[I 210513 17:11:22 _device:721] Download to tmp path: C:\Users\D1M\AppData\Local\Temp\tmp_fig7v5o\_tmp.ipa
Traceback (most recent call last):
  File "d:\python\lib\runpy.py", line 198, in _run_module_as_main
    "__main__", mod_spec)
  File "d:\python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "D:\python\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "d:\python\lib\site-packages\tidevice\__main__.py", line 687, in main
    actions[args.subparser](args)
  File "d:\python\lib\site-packages\tidevice\__main__.py", line 130, in cmd_install
    bundle_id = d.app_install(args.filepath_or_url)
  File "d:\python\lib\site-packages\tidevice\_device.py", line 727, in app_install
    preader.finish()
  File "d:\python\lib\site-packages\tidevice\_utils.py", line 121, in finish
    bytes_per_second = self._copied / (time.time() - self._begin)
ZeroDivisionError: float division by zero

codeskyblue, 2021-05-18:
@a805791737 更新一下tidevice就好了。这个问题既然重复就先关掉了。

closed 2021-05-18 by codeskyblue

xiao-guan, 2023-11-22:
也不是硬要itunes，比如爱思助手也可以。。。在手机上输入密码信任电脑即可

xuyiting777, 2024-12-18:
C:\Users\anker>tidevice list
Traceback (most recent call last):
  File "c:\users\anker\appdata\local\programs\python\python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\anker\appdata\local\programs\python\python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\anker\AppData\Local\Programs\Python\Python39\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "c:\users\anker\appdata\local\programs\python\python39\lib\site-packages\tidevice\__main__.py", line 1027, in main
    actions[args.subparser](args)
  File "c:\users\anker\appdata\local\programs\python\python39\lib\site-packages\tidevice\__main__.py", line 101, in cmd_list
    serial = _d.get_value("SerialNumber")
  File "c:\users\anker\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 366, in get_value
    with self.create_session() as conn:
  File "c:\users\anker\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 312, in create_session
    "HostID": self.pair_record['HostID'],
  File "c:\users\anker\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 137, in pair_record
    self.handshake()
  File "c:\users\anker\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 246, in handshake
    self._pair_record = self._read_pair_record()
  File "c:\users\anker\appdata\local\programs\python\python39\lib\site-packages\tidevice\_device.py", line 173, in _read_pair_record
    data = self._usbmux.send_recv(payload)
  File "c:\users\anker\appdata\local\programs\python\python39\lib\site-packages\tidevice\_usbmux.py", line 54, in send_recv
    self._check(data)
  File "c:\users\anker\appdata\local\programs\python\python39\lib\site-packages\tidevice\_usbmux.py", line 98, in _check
    raise MuxReplyError(data['Number'])
  File "c:\users\anker\appdata\local\programs\python\python39\lib\site-packages\tidevice\exceptions.py", line 25, in __init__
    self.reply_code = UsbmuxReplyCode(number)
  File "c:\users\anker\appdata\local\programs\python\python39\lib\enum.py", line 315, in __call__
    return cls.__new__(cls, value)
  File "c:\users\anker\appdata\local\programs\python\python39\lib\enum.py", line 611, in __new__
    raise ve_exc
ValueError: 22 is not a valid UsbmuxReplyCode