alibaba/tidevice #94: 【HELP】Use it by Java

ZhouYixun, 2021-05-13:
I want to use `tidevice wdaproxy -B com.xxx.WebDriverAgentRunner.xctrunner --port 8200`
before test
and kill it after test.
But I use by
`Runtime.getRuntime().exec('tidevice wdaproxy -B com.xxx.WebDriverAgentRunner.xctrunner --port 8200')`
I cannot kill the process.
If I use .sh to run it,I would not kill three process: relay、xcrun and wdaproxy,because I only known wdaproxy's port.

I wish I can use it in 50+ devices,so I must use it dynamically

jongyu, 2021-05-13:
Use multi-threading and increase available port detection!!!

ZhouYixun, 2021-05-13:
> Use multi-threading and increase available port detection!!!

but I use `tidevice wdaproxy -B com.xxx.WebDriverAgentRunner.xctrunner --port 8200` will increase three random pid,how can I kill them?

codeskyblue, 2021-05-18:
send SIGTERM or SIGINT to the `tidevice wdaproxy`

closed 2021-05-19 by ZhouYixun