alibaba/tidevice #100: 使用文件处理命令 fsync 时增加参数"-B bundle_id"会报错

LRL2018, 2021-05-21:
cmd中运行命名：python37 -m tidevice fsync ls /Documents/PayLoad -B com.ksg.tako
python版本：3.7.9
tidevice版本：0.4.8
报错堆栈如下：
`Traceback (most recent call last):
  File "E:\Python\Python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "E:\Python\Python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "E:\Python\Python37\lib\site-packages\tidevice\__main__.py", line 694, in <module>
    main()
  File "E:\Python\Python37\lib\site-packages\tidevice\__main__.py", line 689, in main
    actions[args.subparser](args)
  File "E:\Python\Python37\lib\site-packages\tidevice\__main__.py", line 384, in cmd_fsync
    pprint(sync.listdir(arg0))
  File "E:\Python\Python37\lib\site-packages\tidevice\_sync.py", line 106, in listdir
    pkg = self._recv()
  File "E:\Python\Python37\lib\site-packages\tidevice\_sync.py", line 75, in _recv
    self.recvall(plist_size) # Discard plist xml-content
  File "E:\Python\Python37\lib\site-packages\tidevice\_safe_socket.py", line 61, in recvall
    raise MuxError("socket connection broken")
tidevice.exceptions.MuxError: socket connection broken`

codeskyblue, 2021-05-21:
看起来连接有点不稳定？？

LRL2018, 2021-05-24:
> 看起来连接有点不稳定？？

我感觉不仅仅是连接不稳定的问题，因为我只要不加 “-B bundle_id"这个选项参数，执行fsync的命令时都是正常输出的。只有在想看某个应用的数据目录时，才会报出上边的错误，请问要查看应用数据目录下的内容是否还需要其他操作？

LRL2018, 2021-05-26:
又试了其他版本的ios设备，我这里的情况是ios14.0以上的版本使用fsync查看应用的沙盒文件就会失败

ashfish, 2021-05-31:
> 又试了其他版本的ios设备，我这里的情况是ios14.0以上的版本使用fsync查看应用的沙盒文件就会失败

IOS14以下正常吗?

LRL2018, 2021-06-01:
> > 又试了其他版本的ios设备，我这里的情况是ios14.0以上的版本使用fsync查看应用的沙盒文件就会失败
> 
> IOS14以下正常吗?

closed 2021-06-01 by LRL2018

LRL2018, 2021-06-01:
> > 又试了其他版本的ios设备，我这里的情况是ios14.0以上的版本使用fsync查看应用的沙盒文件就会失败
> 
> IOS14以下正常吗?

ios14以下是可以的

prophett0527, 2022-11-19:
各位大佬，我在iOS15.7上面也遇到这个问题
tidevice version 0.9.11
新版本 tidevice 有什么进展吗