alibaba/tidevice #102: 【HELP】有没有推荐的分线器？

ZhouYixun, 2021-05-24:
线使用了作者推荐的绿联线，但是连接20台手机还是发生很高的断线率，一晚下来只有一个存活。分线器的话有推荐吗

codeskyblue, 2021-05-24:
我目前用的是这个  https://item.jd.com/4576040.html#crumb-wrap

不过只是因为公司给发的就是这个（其实我个人感觉只要是 USB单独供电的都可以）

codeskyblue, 2021-05-24:
https://item.jd.com/11094070848.html
这个看起来也行（没试过）

ZhouYixun, 2021-05-25:
> https://item.jd.com/11094070848.html
> 这个看起来也行（没试过）

你们一台mac mini能带起多少台ios使用tidevice？我这边带4台以上，就开始各种wda断连的情况，需要额外处理usbmuxd吗？

weikilla, 2021-07-22:
> > https://item.jd.com/11094070848.html
> > 这个看起来也行（没试过）
> 
> 你们一台mac mini能带起多少台ios使用tidevice？我这边带4台以上，就开始各种wda断连的情况，需要额外处理usbmuxd吗？

请问有对比过使用tidevice，会比使用xcodebuild能拖更多的设备吗

slimsymphony, 2021-07-23:
[image]

closed 2021-08-13 by ZhouYixun

codeskyblue, 2021-08-20:
作者很穷的。最近换了一个组，现在能有4台iphone手机就不错了。