alibaba/tidevice #104: 性能采集报错日志

lincanyi, 2021-05-27:
版本
[image]
日志详情：
[image]
报错之后就堵塞住了，无法继续采集

linmscat, 2021-06-16:
这个bug怎么解决啊，遇到了相同的问题 @codeskyblue

linmscat, 2021-06-16:
Traceback (most recent call last):
  File "D:\Users\yl1150\AppData\Local\Programs\Python\Python38\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "F:\GitLab\apprunner\venv\lib\site-packages\tidevice\_perf.py", line 247, in append_data
    for _type, data in idata:
  File "F:\GitLab\apprunner\venv\lib\site-packages\tidevice\_perf.py", line 103, in iter_fps
    for data in d.instruments.iter_opengl_data():
  File "F:\GitLab\apprunner\venv\lib\site-packages\tidevice\_instruments.py", line 797, in iter_opengl_data
    self._reply_null(m)
  File "F:\GitLab\apprunner\venv\lib\site-packages\tidevice\_instruments.py", line 516, in _reply_null
    self.send_dtx_message(m.channel_id,
  File "F:\GitLab\apprunner\venv\lib\site-packages\tidevice\_instruments.py", line 392, in send_dtx_message
    self.sendall(data)
  File "F:\GitLab\apprunner\venv\lib\site-packages\tidevice\_safe_socket.py", line 66, in sendall
    return self._sock.sendall(data)
  File "D:\Users\yl1150\AppData\Local\Programs\Python\Python38\lib\ssl.py", line 1204, in sendall
    v = self.send(byte_view[count:])
  File "D:\Users\yl1150\AppData\Local\Programs\Python\Python38\lib\ssl.py", line 1173, in send
    return self._sslobj.write(data)
ssl.SSLEOFError: EOF occurred in violation of protocol (_ssl.c:2457)
@codeskyblue tidevice.Performance(self._dev)对象调用start()方法会报该堆栈错误

huang1988519, 2021-08-06:
https://github.com/alibaba/taobao-iphone-device/pull/122  尝试合并这个PR再测试是不是一样的问题