alibaba/tidevice #108: 【Want Help】iOS14.6是否无法使用？

VaLueK, 2021-06-03:


AllenLuosong, 2021-06-08:
同问

AllenLuosong, 2021-06-08:
找了一圈，现在没有14.6的开发者镜像,要么回退版本,要么等IOS15

VaLueK, 2021-06-08:
我也找了一圈没找着
https://github.com/iGhibli/iOS-DeviceSupport iOS15的也出了，没出iOS14.6的。我看有条评论下方说下载iOS14.5的改名为iOS14.6可以使用，打算试一下

codeskyblue, 2021-06-08:
稍等一下，马上就好

codeskyblue, 2021-06-08:
DeviceSupport下载地址改成  https://github.com/JinjunHan/iOSDeviceSupport

AllenLuosong, 2021-06-08:
更新了tidevice 0.4.10 再次启动tidevice wdaproxy -B com.facebook.WebDriverAgentRunner.xctrunner --port 8200报错
 RuntimeError: deviceSupport zip not detected DeveloperDiskImage

VaLueK, 2021-06-08:
一样的问题

AllenLuosong, 2021-06-08:
请问这个要怎么处理

codeskyblue, 2021-06-08:
@VaLueK @AllenLuosong iOS版本？

AllenLuosong, 2021-06-08:
IOS 14.6  iPhone12 mini

AllenLuosong, 2021-06-09:
找到解决 deviceSupport zip not detected DeveloperDiskImage 报错的方法了。
**问题回顾**: 最开始我使用iPhone12 mini 版本号14.6 真机调试，运行tidevice wdaproxy -B com.facebook.WebDriverAgentRunner.xctrunner --port 8200 提示 
requests.exceptions.HTTPError: 404 Client Error: Not Found for url: https://github.com/iGhibli/iOS-DeviceSupport/raw/master/DeviceSupport/14.6.zip 这个报错是因为IOS更新到14.6之后tidevice框架没有及时更新依赖的DeviceSupport,
昨天 @codeskyblue 更新了tidevice版本支持IOS14.6之后运行 tidevice wdaproxy -B com.facebook.WebDriverAgentRunner.xctrunner --port 8200  又提示 RuntimeError: deviceSupport zip not detected DeveloperDiskImage 
**解决思路**找到源码如下
    def _request_developer_image_dir(self):
        # use local path first
        # use download cache resource second
        # download from network third
        product_version = self.get_value("ProductVersion")
        logger.info("ProductVersion: %s", product_version)
        major, minor = product_version.split(".")[:2]
        version = major + "." + minor

        mac_developer_dir = f"/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/DeviceSupport/{version}"
        image_path = os.path.join(mac_developer_dir, "DeveloperDiskImage.dmg")
        signature_path = image_path + ".signature"
        if os.path.isfile(image_path) and os.path.isfile(signature_path):
            # yield image_path, signature_path
            yield mac_developer_dir
        else:
            image_zip_path = cache_developer_image(version)
            with tempfile.TemporaryDirectory() as tmpdir:
                zf = zipfile.ZipFile(image_zip_path)
                zf.extractall(tmpdir)
                rootfiles = os.listdir(tmpdir)
                rootdirs = []
                for dirname in rootfiles:
                    if not dirname.startswith("_") and os.path.isdir(dirname):
                        rootdirs.append(dirname)
                if len(rootfiles) == 0: # empty zip
                    raise RuntimeError("deviceSupport zip file is empty")
                elif os.path.isfile(os.path.join(tmpdir, "DeveloperDiskImage.dmg")):
                    yield tmpdir
                elif len(rootdirs) == 1: # contain a directory
                    yield os.path.join(tmpdir, rootdirs[0])
                else:
                    raise RuntimeError("deviceSupport zip not detected DeveloperDiskImage")

这个报错是因为在创建临时文件的时候 没有在临时文件夹中找到 DeveloperDiskImage.dmg，而临时文件夹中的文件是通过本地
\.tidevice\device-support文件中的14.6.zip copy过去的，tmpdir中就只有一个14.6文件，DeveloperDiskImage.dmg在14.6文件中的下一层。
解决方法，1.修改源码查找DeveloperDiskImage.dmg的逻辑，查找tmpdir的14.6文件夹
2.将14.6.zip中的DeveloperDiskImage.dmg复制到和zip中的14.6文件夹的同一目录中

codeskyblue, 2021-06-10:
@AllenLuosong 看你的描述，应该就是解压后的tmpdir下有两个目录。其中有一个14.6的文件夹。我这样处理了一下

```python

codeskyblue, 2021-06-10:
@AllenLuosong @VaLueK 试试看是不是恢复了

VaLueK, 2021-06-15:
我这边ok~