alibaba/tidevice #109: 求助 没xcode,可否给个wda.ipa

at89s52rc, 2021-06-03:
哎,没台mac,太惨了,连wda都搞不了

codeskyblue, 2021-06-08:
This is the old version of WDA
https://github.com/alibaba/taobao-iphone-device/blob/f7ea95d3f9edc3a60950fbc5c3cefe07105e419c/tests/testdata/WebDriverAgentRunner.ipa

linmscat, 2021-06-23:
> This is the old version of WDA
> https://github.com/alibaba/taobao-iphone-device/blob/f7ea95d3f9edc3a60950fbc5c3cefe07105e419c/tests/testdata/WebDriverAgentRunner.ipa

(venv) F:\GitLab\apprunner>tidevice install WebDriverAgentRunner.ipa
Copying 'WebDriverAgentRunner.ipa' to device... [8.5 MB/s 1s] DONE.
Installing 'com.facebook.WebDriverAgentRunner.xctrunner'
- CreatingStagingDirectory (5%)
- ExtractingPackage (15%)
- InspectingPackage (20%)
- TakingInstallLock (20%)
- PreflightingApplication (30%)
- InstallingEmbeddedProfile (30%)
- VerifyingApplication (40%)
[E 210623 15:48:08 _device:740] ApplicationVerificationFailed
[E 210623 15:48:08 _device:741] Failed to verify code signature of /var/installd/Library/Caches/com.apple.mob
ile.installd.staging/temp.GUqDIu/extracted/Payload/WebDriverAgentRunner-Runner.app : 0xe8008015 (A valid prov
isioning profile for this executable was not found.)
Traceback (most recent call last):
  File "D:\Users\yl1150\AppData\Local\Programs\Python\Python38\lib\runpy.py", line 193, in _run_module_as_mai
n
    return _run_code(code, main_globals, None,
  File "D:\Users\yl1150\AppData\Local\Programs\Python\Python38\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "F:\GitLab\apprunner\venv\Scripts\tidevice.exe\__main__.py", line 7, in <module>
  File "f:\gitlab\apprunner\venv\lib\site-packages\tidevice\__main__.py", line 712, in main
    actions[args.subparser](args)
  File "f:\gitlab\apprunner\venv\lib\site-packages\tidevice\__main__.py", line 131, in cmd_install
    bundle_id = d.app_install(args.filepath_or_url)
  File "f:\gitlab\apprunner\venv\lib\site-packages\tidevice\_device.py", line 742, in app_install
    raise ServiceError(progress['Error'])
tidevice.exceptions.ServiceError: ApplicationVerificationFailed
用你给的这个IPA安装，会提示这个错误，请问如何解决？ @codeskyblue

codeskyblue, 2021-06-30:
@linmscat 不重签名是不行的

jeniva, 2021-08-19:
Windows上有什么方便重签名的办法吗？

nanoscopic, 2022-04-01:
In order to re-sign anywhere you first need to acquire your signing credentials. I don't know any way to do that except using a MacOS instance. I would suggest spinning up MacOS in a VM first to extract your signing credentials. The credentials can then be used with various re-signing tools to do the re-signing...