alibaba/tidevice #116: WDA is not response| WDA stopped | UsbmuxReplyCode.ConnectionRefused

smilemoon777, 2021-06-16:
> @name-jiang 你是在测试什么应用，我昨天试了下抖音这种应用点击也会卡死掉。不是tidevice故意的，是WDA不响应了。

我也发现不是很稳定，出现一次与wda的连接超时，就会有一段时间无法使用，tidevice就偿试重连，过段时间就又可以了。

帖一下tidevice的日志：
[I 210611 14:21:58 _device:1003] Start execute test plan with IDE version: 29
[I 210611 14:21:58 _device:913] WebDriverAgent start successfully
[D 210611 14:22:31 _wdaproxy:93] [f26812b84ac67cc567707cb6a679f8ba37095936] WDA is not response, check again after 10s
[D 210611 14:24:16 _wdaproxy:93] [f26812b84ac67cc567707cb6a679f8ba37095936] WDA is not response, check again after 10s
[I 210611 14:24:29 _wdaproxy:97] [f26812b84ac67cc567707cb6a679f8ba37095936] WDA confirmed not running
[I 210611 14:24:29 _wdaproxy:106] [f26812b84ac67cc567707cb6a679f8ba37095936] WDA keeper stopped
[I 210611 14:24:29 _wdaproxy:145] [f26812b84ac67cc567707cb6a679f8ba37095936] WDA stopped for the 19st time, running 2.5 minutes
[D 210611 14:24:29 _wdaproxy:118] [f26812b84ac67cc567707cb6a679f8ba37095936] launch WDA
[I 210611 14:24:30 _device:956] BundleID: com.facebook.WebDriverAgentRunner.xctrunner
[I 210611 14:24:30 _device:959] ProductVersion: 12.1

_Originally posted by @yangeren in https://github.com/alibaba/taobao-iphone-device/issues/79#issuecomment-859311742_

zealotZB, 2021-06-23:
我也出现了，直接启动facebook-wda，说wda在后台运行，但是就是不响应，但是一会又可以 一会又不可以

WindyTouch, 2021-06-28:
我也出现了这个问题，而且感觉出现的还挺频繁，经常报not response

codeskyblue, 2021-06-30:
查了一下代码是有点问题，之前检查的间隔太短只有3s，而滑动这类操作是远超3s的。这次修改将检查间隔调整为30x2=60s。版本0.4.12
另外提供的关闭检查的方法（虽然我不是很建议）
`tidevice wdaproxy --check-interval 0`

zhixideyu, 2021-09-28:
> 我也出现了，直接启动facebook-wda，说wda在后台运行，但是就是不响应，但是一会又可以 一会又不可以

你好，这个问题现在有什么进展吗

weiwei-huhu, 2023-05-12:
hello，这个问题有什么进展吗？